foldynl/QLog #407: New Window "Details"

OZ1CT-Ben, 2024-07-07:
would it be possible tom make a new "Windows" whit a "Details" ???

I think the info in "Details" is somthing you use when you setup the log, so my idea was to make it turn on/off if you need it. for my reason it will give me more space on the dashboard for Cluster-Rotor-Chat to use this place ???

[image]

73" OZ1CT Ben

foldynl, 2024-07-08:
Unfortunately, getting possible to switch on/off this window is not as simple as it might seem at first. It is part of the NewContact window. This would mean redesigning the entire NewContact function, which would also have a negative impact on Layout Settings and other parts of QLog. The negatives outweigh the positives of the change.

OZ1CT-Ben, 2024-07-08:
Ok...then this idea is grounded 😉

73" Ben

Send from my mobile office.

closed 2024-07-10 by foldynl