foldynl/QLog #408: Layout - menu button ??

OZ1CT-Ben, 2024-07-07:
My idea was to quick and easy change the Layout of your dashboard, if you change from "HF-Phone" and go to "QO-100" just hit the menu-button and all is change for you....also the Rig setup/rotor/....
there is space on the dashboard for it on place 1. or 2.

73" OZ1CT Ben
[image]

foldynl, 2024-07-08:
basically it's a good idea. It is necessary to improve it. It is not possible to use buttons here, because the operator can define several profiles, which would mean that the QLog layer would be full of buttons. But there is definitely a need to do something with the fact that Qlog contains several profiles that can be switched independently. I am registering a request to switch the antenna profile depending on the band and rig. And this idea could solve it. Thanks

OZ1CT-Ben, 2024-07-08:
Is it maybe better to have a startup menu and here select where you will be
active (HF-VHF) and then it start up in layout you "made" ???
Send from my mobile office.

foldynl, 2024-11-24:
The new release 0.40.0 contains the Activity Manager

closed 2024-11-24 by foldynl