foldynl/QLog #411: clera Gridsquar when callsign is written

OZ1CT-Ben, 2024-07-10:
Hi Ladislav

if you hear a weak station and got his "Gridsquar" and type it in, but when you start to 
write his call in "Callsign" it wipe out all info in the "Gridsquar" ????

this occurs when I did a qso on a rover station but in a new grid square on qo-100

73" Ben

foldynl, 2024-07-10:
Yes, this feature has been there almost since the beginning. I don't know what solution you would suggest, but it is necessary to look at it from a different point of view than the one you describe in your case. There are many other cases where leaving the gridsquare or other fields would be unwanted. QLog cannot meet all of these requirements, and a compromise must always be chosen. For example, in a situation where you enter a callsign and the callbook's data are used, you might find out that the callsign is wrong. Yes, it is possible to press Esc, but it is faster to delete the QSO fields automatically when changing the callsign.

I am not against change, but I would like the change to be thought out from all angles and not just for one specific case.

OZ1CT-Ben, 2024-07-10:
Hi Ladislav

Yes I can see if you have the QRZ or other online database to seek for the
grid squar you have to make a choice.....so I think I keep the a paper on
the desk and then write a note ;-)

73" Ben

foldynl, 2024-07-10:
Sometimes it's hard to please everyone. Making a change is not a problem, but imagining all the consequences that arise from it can be challenging. Yes, I have encountered this problem before. However, I don't have a good solution now. As I mentioned, it is possible to turn this off and let the operator use ESC for deletion. But there are many QSO fields and it is possible to sometimes overlook something when callbook data are used. That's why there is this feature. In the other word, It is designed for the work-flow: catch a callsign and then add the details.

closed 2026-09-18 by foldynl