foldynl/QLog #416: Callbook data collection

dl2ki, 2024-07-19:
Today I transferred the callsign "G4WQI" from the band map to the NewQSO window. I noticed that the Gridsquare field was empty. In my settings I had "QRZ.com" as the primary callbook and "HamQTH" as the secondary callbook.

[image]

After reversing the callbook order in the settings, the Gridsquare field was filled with "JO01HT".

[image]

So I looked at both callbook pages and found that Gridsquare was specified in both callbooks.

[image]  [image]

[image]

I cannot say, how the internal data query of the callbooks in QLog works and which data the respective APIs provide. In any case, the data source can be easily distinguished by the capitalization of the name in QRZ.com.

On the other hand, if two callbooks are set up, you could query the data from both callbooks and add/combine the missing data in the primary callbook for the NewQSO window from the secondary callbook if it is available there.

Of course, there may be other side effects that prevent this.

Now, here I will use HamQTH as the primary callbook.

foldynl, 2024-07-20:
Wolfgang, if you are using QRZ.com and  want to get all information (including gridsquare), you need a QRZ subscription. Unfortunately, It is QRZ's policy. If you have no subscription, then you get a limited set of information where the gridsquare is not included. So the QLog behavior is correct in this case.

the algorithm for obtaining information from the callbook is not very simple, because prefixes and suffixes, but simply algorithm is: If the callsign is found in the primary, use all available information. If the callsign is not found in the primary, then the secondary is used. Information are not combined .

The question about combination of information from several callbooks. I'm not sure about that, because both places may have different updated data - someone just tried HamQTH but stayed to QRZ. Then, you can't combine these data, because you don't know how up-to-date they are. I think that obtaining and using information from one source is more reliable than a combination of two sources that may be different up-to-date.

dl2ki, 2024-07-20:
If QRZ.com does not provide the Gridsquare data, that's OK. Combining data from different sources always raises the question of data consistency. So I can understand your concern. In such cases I use Gridsquare from the details tab, which is no problem at all.

In the meantime I have set HamQTH as the primary callbook. This can also be used to support Petr's work.

closed 2024-07-24 by foldynl