foldynl/QLog #417: Set rotator position fails 40% of the time

oh2gba, 2024-07-19:
version: 0.37.1, on Ubuntu 22.04.4

About 40% of the time, setting the position of the rotator fails with the following warning pop-up:

[image]

Whereafter the rotator needs to be re-connected using the menu 'Equipment'

Hereby the log file, where the event happens: [qlog_rot_issue.log](https://github.com/user-attachments/files/16307275/qlog_rot_issue.log)

And for completeness the configuration:
[image]

Thank you for your effort into this software!
73, John de OH2GBA

foldynl, 2024-07-20:
Thanks for the log. Please, could you send me a log where set position is successful ? Unfortunately, it doesn't seem like it is a QLog error, because Qlog sent a request to set AZ and got a response invalid command from Hamlib. But the same command is sent every time you want to set AZ.

oh2gba, 2024-07-20:
Thank you for your reply, please find attached a log, from program start, rotator fully working, it receives the position correctly, and set's it once correctly. 

[qlog_rot_issue_2.log](https://github.com/user-attachments/files/16319656/qlog_rot_issue_2.log)

However after trying a bit more I may have a small clue, 

Succesful steps:
1) changing the rotator position just by clicking the world map on the rotator-window. 
2) successful whatever I tried, 

How to Fail it:
1)  clicked a station from the DX-Cluster, 
2) hit the "QSO-SP" button on the rotator-window, 
Result: fails immediately.

foldynl, 2024-07-20:
Do the examples above go to the error/success ? That means, is it 100% replicable in the given cases?

I would like to understand your second qlog_rot_issue2.log file. From the log, you requested an azimuth 93. Then I can see that the antenna move from azimuth 4 to azimuth 0 where stopped. Are you using some antenna offset or why the antenna stopped at azimuth 0?

foldynl, 2024-07-20:
Finally,  it's a QLog's issue. I misunderstood the Hamlib library. I tested everything against the old Hamlib's Dummy Rotator, which has an azimuth interval from -180 to 180. I mistakenly thought that hamlib recalculates to this interval. So QLog recalculates everything to the interval -180 to 180. However, this is only in the case of old Dummy Rotator. All others have an interval of 0-360 or greater. Therefore, it should only be a problem if you require beaming to azimuths 180 or more.

oh2gba, 2024-07-20:
Hereby new logs, both the Qlog and a serial spy log. Serial spy shows the data between my computer and the rotator. 

Several times the position has been sent correctly, and 2 times it failed, where after I restarted the communication 
by enabling the rotator again in the "Equipment" menu. 

Hope this helps. 
[qlog_rot_issue_3_qlog.log](https://github.com/user-attachments/files/16321170/qlog_rot_issue_3_qlog.log)
[qlog_rot_issue_3_serial.log](https://github.com/user-attachments/files/16321171/qlog_rot_issue_3_serial.log)

There is no offset defined for my rotator. Also I can not figure 100% accurate steps to reproduce this.

oh2gba, 2024-07-20:
Ah no I see your other post, yep you may be right, anything above 180 fails :)

commit 3f50b89e (2024-07-26): Fixed #417 - Set rotator position fails if azimuth > 180
closed 2024-07-26 by foldynl