foldynl/QLog #418: Window State Not Persisted After Restart: Fails to Remember Full-Screen and Window Sizes

oh2gba, 2024-07-20:
This is my screen before closing:
[image]

Close Qlog and reopen it again, the size of the main window is smaller now, and I have to maximize it
[image]

After maximizing, the window sizes are not the same anymore, see dx-cluster and online-map are smaller now. 
[image]

It would be _so_ nice if Qlog would just remember all sizes! Perhaps not a bug, but a kind request :)

John

dimarmnes, 2024-07-20:
I agree with John's request.

v037.1 on Manjaro Linux

73
LU1IDC, Diego

foldynl, 2024-07-20:
I know about the issue but I don't know how to fix it. It occurs only in case when the full screen layout is saved.

dl2ki, 2024-07-22:
Suggestion:

If a solution can be found here, perhaps the position and zoom factor of the band map could be saved at the same time, so that they are restored when QLog is restarted.

On the other hand, there may be reasons why other settings or no settings at all for displaying the bandmap when starting QLog may make more sense.

dl2ki, 2025-01-22:
System: Kubuntu 24.04.01 / QLog 0.41.1

I have noticed that the dimensions of the individual windows have changed after restarting QLog. The settings for the window widths and heights are apparently not saved.

This issue indicates that this is only the case in QLog fullscreen mode. I don't use QLog in fullscreen mode because I still need access to some desktop elements on the side of the QLog window.

I have therefore reduced the window size once, to show a maximum effect,  and then closed QLog.

Here are two pictures. Both layers have been adjusted to 1908 x 1064 px for comparison.

**555 x 869 px**
[image]

After restarting you can see that the window sizes are not the same as before.
 
**1908 x 1064 px**
[image]

If I close QLog in the size used here, the overall size of the QLog desktop does not change when I restart it. Only the individual settings of some window widths and heights change.

dl2ki, 2025-01-22:
OK, thanks for the hint.

In ‘Classic’ mode, restoring the set window sizes works. However, the display of ‘TX PWR’ is missing here.

In ‘Activity’ mode (here: dl2ki), ‘Save Arrangement’ must always be pressed after setting the window sizes.

When restarting, however, other window sizes continue to appear. However, the previously set window sizes appear when I switch the activity from ‘dl2ki’ to ‘classic’ and then back to ‘dl2ki’.

dl2ki, 2025-01-24:
Hi,

I have experimented a little here. The behaviour with the required double switching of the activity can certainly be reproduced here. Is this also the case for you, or could there be a cause outside of QLog?

foldynl, 2025-01-24:
Wolfgang, to be honest, I do not really understand your issue. Is the problem that the window size changes between restarts?

I tried both full-screen and non-full-screen modes, and if the settings are saved, everything returns to the default size after both restarts and profile switches. Even when switching back to the same profile, everything is set to the correct size. Overall, the issue is that there is a bug in Qt, and QLog implements some workarounds to compensate for it. Maybe there is the issue.

The important thing is to press Save Arrangement when you are satisfied with the window layout. I will try to run some additional tests over the weekend.

dl2ki, 2025-01-24:
Hi Ladislav,

I have sent you a demo video by e-mail. Perhaps the behavior of QLog is now a little clearer.

commit 75c4f70b (2024-07-23): Fixed #418 - Windows State/Size does not save in case of fullscreen (2)

QApplication::desktop is obsoleted and QT6.x does not implement it.
This commit uses QGuiApplication::screens
commit ff9449af (2024-07-23): Fixed #418 - Windows State/Size does not save in case of fullscreen (3)

Need to apply only for QT under 6.3. Otherwise, other QT issue is present.
commit 7149a0c9 (2024-07-26): Fixed #418 - Windows State/Size does not save in case of fullscreen (4)

If it was like this, then the QT error occurred because no windows frame
was present when switching to the same profile.
commit a7ed96c9 (2024-07-26): Fixed #418 - Windows State/Size does not save in case of fullscreen

It seems to be a QT issue. The workaround described in the issue
QTBUG-46620 was applied, but not everywhere restoreGeometry is called, as it caused
other issues in QT, which are also detailed in the bug report. While it works better,
it is not perfect.

Bug QTBUG-46620 was fixed in QT 6.3.0, so it is necessary to determine how this workaround
behaves under this QT version. This could affect Windows users.
closed 2025-02-14 by foldynl