foldynl/QLog #425: SIG field dropdown list

dimarmnes, 2024-07-24:
I often use the SIG field to log activations of lighthouses, museums, train stations, campsites, etc. 
To ensure consistency in naming and make searching easier over time, I believe it would be helpful to select from a predefined list. Ideally, this would be a user-generated dropdown list where individuals can add their preferred SIG categories.

Diego, LU1IDC

foldynl, 2024-07-25:
SIG is defined as a Free-form field in the ADIF Specification. So it is not possible to define a fixed list of values. An user pre-defined list is not the way I would like to see it, as this complicates the QLog setup.

What I would like to see and it is a question for you, whether it would be possible to use Completer as it is used in the case of SOTA or POTA. The values, which are already from past QSOs, would then be used. If you wanted a new one, you would write it down. Would it be possible?

dimarmnes, 2024-07-25:
That would be a valid solution of course.

foldynl, 2024-08-29:
PR is part of v0.38