foldynl/QLog #436: Added SWL Check field

foldynl, 2024-08-01:


foldynl, 2024-08-05:
Maybe I was wrong, and ADIF defines SWL as a field indicating that the given record was made by an SWL station. May I ask how you, @ik3ocd and @AsciiWolf, envision the SWL-check field function? What should it be used for?

ik3ocd, 2024-08-05:
good morning
Ladislav, now for the swl field, in my practical thought it should be used
for a possible response to an swl both as a card and as an e-qsl, sometimes
I observe on the -eqsl site swl requests to confirm their listening to my
callsign and I do it manually, this happens because in the e-eql reception
analysis they do not consider the qsos not inserted in my log in this the
swl and the qsos that did not go well but that the correspondent wanted to
send me the e-qsl, I do not want to create unnecessary practical problems
but according to my thought qlog should import and save separately a file
in adif of the qsos not present in the personal log during the loading
phase of -eqsl, to then be able to analyze and possibly import the qsos and
also those of the swl and then subsequently respond. here Ladislav I gave a
suggestion on the swl service then the question is all yours obviously the
programmer is you and you think what is more correct or not. best regards
and thanks for the work done qlog.
ps:
I note that during the e-qsl qlog loading phase it establishes that there
are qsos not present in the log, and these are the ones that should be
recovered. Among these there could be an swl that requires confirmation.

foldynl, 2024-10-19:
I'm just attaching a link to the ADIF discussion, which shows that the ADIF specialists don't know how exactly SWL QSOs should be recorded - BTW, it is a long email communication

https://groups.io/g/adifdev/topic/109062618#msg9817

ik3ocd, 2024-10-19:
hello ladislav thanks for the information, i understand, and i read that
there is a certain sadness of some om towards the swl, i say this because i
too was a radio listening enthusiast and how i wanted some answers from a
radio amateur, i leave the decisions to the programmers that i thank for
their work and activity in the radio world. for the rest i will answer
manually through the e-qsl site to a possible request from a swl. best
regards.

foldynl, 2025-07-11:
Unfortunately, there is no support for it in ADIF - closing this

closed 2025-07-11 by foldynl