foldynl/QLog #438: possibility to vary the character size in DX-Cluster window

ik1vqy, 2024-08-04:
Goodmorning and thanks for Your work.

It would be nice to be able to vary the font size in the cluster window... Age advances...Hi!

Thanks
Alfredo IK1VQY

foldynl, 2024-08-04:
Hi,

it is already implemented. `CTRL` + `Mouse Wheel`. See [QLog Wiki Shortcuts](https://github.com/foldynl/QLog/wiki/Shortcuts) for more details

ik1vqy, 2024-08-04:
Ooops, I haven't seen it, but I ment in the spot window of the cluster.  In that view the font size doesn't increase with control + weel of the mouse. I hoped in this possibility
Thanks
Alfredo

foldynl, 2024-08-05:
Unfortunately, it is not currently planned to have the possibility to change the font size for the selected window. If it needs to be set, it must be set for the entire application and not just for DXC. However, as a result, the geometry of the windows may be disturbed.

If you need a larger font size, it may affect the display settings of the application in the OS. For example, in Linux, it is called Scale Factor, and under Windows, you can find it in the Screen settings.

closed 2024-08-29 by foldynl