foldynl/QLog #441: Owner Callsign, Station Callsign, Operator

GregTroar, 2024-08-14:
Hello,

Just discovered this log and trying it.
One of my issue is that in the settings / station we cannot put the Owner callsign as well as Operator callsign, it seems we can only put station callsign.

Same for the filters I do not find these ones.

Thanks

foldynl, 2024-08-16:
QLog was originally designed as a 'personal' log application. This means that these fields in the Station profile are currently deliberately missing. Unfortunately, it's not just about adding fields to the Station profile, but also about changing the logic in TSQL and other exports.

Could you please describe a use case for why you need these fields? Are you a club station?

Regarding filters, it is possible to create a user-defined filter for each item in the log.

GregTroar, 2024-08-16:
Hi,

I have two calls here in Vietnam, XV9Q & XV9BPO that I have to use depending on my location or TRX I use.
Owner call is always XV9Q as my main call, and station callsign is either XV9Q or XV9BPO while operator can be different in case a friend use the station under my call.
Finally it is also to keep those fields filled in case of export to other log which manages those fields and for filter purpose.

Anyway if this cannot be changed no big deal, I was just testing QLog which seems to be very nice and working well

foldynl, 2024-08-16:
thanks for the description. Something from your description can be solved by switching the Station profile. However, not all fields you wrote above will be filled. At present, however, it is not possible to correctly set the case when your friend use your callsign.

I found probably the best description of how these fields should be used on the [ADIF mailing list](https://groups.io/g/adifdev/message/8861)

dimarmnes, 2024-08-16:
Hi Ladislav,
What do you think about adding the Operator field to the My Station tab in
the New QSO window?
Although I haven't had to use it yet, I may also find myself in the
situation where another Callsign/Operator uses my station; in that case I
would like and should leave the Operator Callsign recorded.
I understand that the logbook belongs to the station and not the person. I
don't consider it critical but I think it would be a nice addition.
Diego Carratini

closed 2024-11-24 by foldynl