foldynl/QLog #443: [Question] eQSL QTH nickname and duplicated QSOs

jrbenito, 2024-08-19:
Hi,

Thanks for this logging program. I am facing an issue, probably my fault, but I can't figure it by myself.

First, when uploading to eQSL.cc, no matter what I put in the QTH nickname, I always get an error:

[image]

I´ve tried change it on eQsl website, but no luck. Hence, when I try to download confirmed QSOs, I never get any.

Second, and I think both issues are related, when I send QSOs without specifying the Nickname, I got them on both profiles I have in eQSL.cc (VE7LBD - my default profile and call sign - and also PU2LBD my Brazilian call sign). The eQSL website will say I have invalid QSOs on my PU2LBD profile and on VE7LBD it will tell "dupe found PU2LBD".

I do not have the PU2LBD QTH setup in Qlog, this callsign has been inactive for while. However, if I do not specify QTH nickname on QLog and try to download QSO confirmation from the time I was operating under PU2LBD (but specifying VE7LBD in Qlog), them it downloads QSOs confirmation from PU2LBD and messages me that there is not correspondence in the logbook (because this is only for VE7LBD).

What I am doing wrong?

Thanks a lot for your support.

foldynl, 2024-08-20:
Please, could you send me a screenshot of the eQSL Setting Dialog with your current setting? Please, could you export one QSO what you want to upload to eQSL and send me it ? QLog does not use APP_EQSL tag. It seems that you have imported QSOs with this tag.

jrbenito, 2024-08-20:
So, starting from eQSL.cc:

[image]

No matter how I call the QTH in eQSL.cc it never wrorks when I put them onto QLog (image on previous post).

[image]

Now, the upload from QLog to eQSL.cc works if I don't specify a QTH nickname:

[image]

The confirmation download never works as shown in the previous post. However, on eQSL.cc I have all duplicated:

[image]

and

[image]

But on VE7LBD I also got those (as expected):

[image]

And finally a QSO exported from QLOG

[qlog.adi.zip](https://github.com/user-attachments/files/16679066/qlog.adi.zip)

And same qso on eQSL

[image]

If you need any more information, please let me know.
Thanks!

jrbenito, 2024-08-21:
@aa5sh 

So my assumption is, since I cannot specify any QTH nickname on QLog (any I put fails), I am uploading without specifying it. And since I have more than one account, it is making eQSL.cc confusing the QTHs.

If this holds true, it is all related to why I cannot specify the QTH name?

aa5sh, 2024-08-21:
That first message you posted makes me think that qLog is sending it but eQSL don't like it for some reason but I could be wrong.  I don't have multiple locations in eQSL I can try to add one to test. I struggled with it a while back.  I have two station locations I operate from and I tried to create them but had problems with times overlapping.  I at times will operate both at the same time/weekend and eQSL looks like I would need to create entries for every flip.  But I may be wrong.

aa5sh, 2024-08-21:
This message here is showing the error that was received back from EQSL.cc 
[image]

aa5sh, 2024-08-21:
Ok I was just playing with it here.  This is how my eqsl.cc looks 
[image]
For me to download QSLs from my old callsign I need to change my eqsl.cc login to my old call for instance n5omp with my current password and it downloaded those entries.  Maybe try to upload/download using your ve7lbd call and see what happens.

jrbenito, 2024-08-22:
> or me to download QSLs from my old callsign I need to change my eqsl.cc login to my old call for instance n5omp with my current password and it downloaded those entries. Maybe try to upload/download using your ve7lbd call and see what happens.

That worked - and that was the only thing I did not try (shame on me).

That will make it for now. Thanks.

jrbenito, 2024-08-22:
@foldynl if you want to close this ticket since it seems more like a eQSL thing and not QLog...

Thanks for supporting.

foldynl, 2024-08-22:
@aa5sh thanks for testing. 

So the lesson learned from this is that it is always necessary to enter a callsign for which a nickname is defined. Is that so?

Another observation for improvement is that it is necessary to have a profile for external services (eQSL, lotw etc.) defined for each Station profile.  That sounds pretty complicated for a common operator.

aa5sh, 2024-08-22:
Yeah I think so.  So always use the callsign as the username.  Then you can
use the nickname if you have multiple entries for the same callsign.  So if
I was to move I would put in a end date and a new entry with new start date
and a nickname.  Then I would start adding the nickname.  Although Eqsl.cc
says they will first match off callsign and date.  Then look at the
nickname.  So I really don't understand the value of the nicknames.  Since
they will not let you have times overlap for the same callsign.

[image: image.png]

jrbenito, 2024-08-22:
> I really don't understand the value of the nicknames. Since they will not let you have times overlap for the same callsign.

I agree, for me, the idea of multiple QTH is to be accurate when operating portable, pota or having a vacation home somewhere in another grid, maybe another country.

My case, I intend to operate PU2LBD next year when vacationing at my mom's house and also from the beach there, two QTHs. Here in Canada I intend to go POTA around and my brother is in VE6, hence another portable variation of my QTH. If the QTH dates cannot overlap, it makes little to no sense having them.

aa5sh, 2024-08-22:
Yeah the way I understand it, you would need to put start/stop dates to
break them all up.  I am fortunate to have stations at two locations.  I
may use one tonight and the other tomorrow.  No easy way to identify them
with the way it is currently implemented at eQSL.

foldynl, 2026-10-28:
>>> if you want to close this ticket since it seems more like a eQSL thing and not QLog...

I am closing this.

closed 2026-10-28 by foldynl