foldynl/QLog #452: Awards / 60m?

3JWdotcom, 2024-09-04:
It would be nice to find the 60m Band also in the Awards section.

73s Dirk

foldynl, 2024-09-04:
it doesn't make sense for [DXCC](http://www.arrl.org/dxcc-rules), WAC, [WAZ](https://www.google.com/url?sa=t&source=web&rct=j&opi=89978449&url=https://www.k0nr.com/wordpress/wp-content/uploads/2024/02/cq_waz_rules_english.pdf&ved=2ahUKEwjFqbGY-aiIAxWLxQIHHaV6AOEQFnoECBcQAQ&usg=AOvVaw0sv4-qYZyD0--yHCdJmC2i) and other awards because of the rules (see links). It makes sense to POTA, SOTA, WWFF only

3JWdotcom, 2024-09-04:
and for my personal ;-) statistics how many DXCCs are worked/cfm already, sure for the official awards not, but.....it would be a nice2have. And...who knows maybe in another couple of years!?

kind regards
Dirk

commit ab9e962a (2024-09-20): Fixed #452 - Bands are displayed based on the Settings
closed 2024-10-05 by foldynl