foldynl/QLog #453: Click on PHONE DX Spots sets wrong mode

foldynl, 2024-09-04:
From mailing list:

Interesting phenomena with the newest release when using with the Yaesu FTDX101MP.

Never experienced this before, however now when I click on a DX spot, and the rig switches to the band and frequency, often the sideband (USB / LSB) is not correct for the band. For example, if I click on a spot from 20 meters, it often switches the rig over to LSB. On 40 meters often to USB. If I clink that spot several times, Qlog finally switches to the correct sideband.

foldynl, 2024-09-04:
According to the test, it is not clear whether it is a Hamlib or Qlog error. Based on a the QLog's trace file, it seems that Qlog sends the correct mode, but Hamlib reports other mode
[qlog.trace.txt](https://github.com/user-attachments/files/16870384/qlog.trace.txt)

Tested also with current Hamlib master branch

jmeloranta, 2024-10-05:
and sometimes this leads to leaving the mode to CW. Then I have to manually switch to USB (or LSB).

foldynl, 2025-04-05:
Is this issue still present ?

aa5sh, 2025-04-05:
I just tried it here with my FTDX-101D and it still occurs.  I was on 20M CW and double clicked on a spot that should have brought me to 10M SSB 28.462 and it stayed on CW. If I double click again on the spot to try and have it switch to SSB it will switch to SSB but the frequency shifts down some appears to be the difference in my cw side tone to 28.461.620.  Double click again and it is good.

I am using HamLib connecting over RigCtlD.

aa5sh, 2025-04-05:
[qlog_2025-04-05.trace.txt](https://github.com/user-attachments/files/19617843/qlog_2025-04-05.trace.txt)

I just grabbed this trace.  I tried to go from 10M USB to a spot on 17M CW.

foldynl, 2025-04-07:
Thanks for the detailed log. Unfortunately, what Rig is reporting doesn't make sense to me. I've listed the important things here.

Summary:
- Qlog sent a mode change request 'CW'
- Rig reported a mode change 'CW'
- QLog sent a freq change request '18.07310'
- Rig reported a freq change '18.07310'
- Rig reported a mode change 'USB' - Filter Width 2100Hz -- WHY !? I cannot explain it
- QLog sent a mode change 'CW' after 6 second (maybe the second click)
- Rig reported a mode change 'CW' - Filter Width 2100Hz
- Rig reported a freq change '18.07368' - WHY ? I cannot explain because there was no request from QLog
- Rig reported a filter width change from 2100Hz to 600Hz - mode remained unchanged 'CW'

So the questions are:
- why did rig change the mode from CW back to USB  - it was not requested by QLog
- why did rig change the frequency to 18.07368 - it was not requested by QLog
- why the filter was changed form 2100Hz to 600Hz - it was not requested by QLog
 
I can't explain the questions from the log and I need your help. Somehow debug it because these changes were not requested from QLog.

I'm going through the FTXD Manual to see if I can find anything. There might be some automatic action described there.

For better tracking, I'm attaching fragments from the QLog log where the changes are reported. It's important to note that QLog periodically reads the Rig's status. Requests to the Rig are sent asynchronously, but there's a central lock to prevent simultaneous read/write operations. Finally, there's a 200ms delay between commands (read/write) to prevent a rig "overloading"

- QLog sent a mode change CW
```
13:28:02.468 [DEBUG   ] [0x16bbcf000] [qlog.rig.driver.hamlibdrv.runtime]                Get PWR is disabled [void HamlibRigDrv::checkPWRChange():../rig/drivers/HamlibRigDrv.cpp:770]
13:28:02.468 [DEBUG   ] [0x16bbcf000] [qlog.rig.driver.hamlibdrv.function.entered]       *** [void HamlibRigDrv::checkRITChange():../rig/drivers/HamlibRigDrv.cpp:775]
13:28:02.468 [DEBUG   ] [0x16bbcf000] [qlog.rig.driver.hamlibdrv.runtime]                Get RIT is disabled [void HamlibRigDrv::checkRITChange():../rig/drivers/HamlibRigDrv.cpp:827]
13:28:02.468 [DEBUG   ] [0x16bbcf000] [qlog.rig.driver.hamlibdrv.function.entered]       *** [void HamlibRigDrv::checkXITChange():../rig/drivers/HamlibRigDrv.cpp:832]
13:28:02.468 [DEBUG   ] [0x16bbcf000] [qlog.rig.driver.hamlibdrv.runtime]                Get XIT is disabled [void HamlibRigDrv::checkXITChange():../rig/drivers/HamlibRigDrv.cpp:884]
13:28:02.468 [DEBUG   ] [0x16bbcf000] [qlog.rig.driver.hamlibdrv.function.entered]       *** [void HamlibRigDrv::checkKeySpeedChange():../rig/drivers/HamlibRigDrv.cpp:889]
13:28:02.468 [DEBUG   ] [0x16bbcf000] [qlog.rig.driver.hamlibdrv.runtime]                Get KeySpeed is disabled [void HamlibRigDrv::checkKeySpeedChange():../rig/drivers/HamlibRigDrv.cpp:917]
13:28:02.850 [DEBUG   ] [0x1f7b4c840] [qlog.rig.rig.function.entered]                    *** [void Rig::setMode(const QString &, const QString &, bool):../rig/Rig.cpp:417]
13:28:02.850 [DEBUG   ] [0x1f7b4c840] [qlog.rig.rig.function.parameters]                 "CW" "" false [void Rig::setMode(const QString &, const QString &, bool):../rig/Rig.cpp:419]
13:28:02.850 [DEBUG   ] [0x16bbcf000] [qlog.rig.rig.function.entered]                    *** [void Rig::setModeImpl(const QString &, const QString &, bool):../rig/Rig.cpp:431]
13:28:02.850 [DEBUG   ] [0x16bbcf000] [qlog.rig.rig.function.parameters]                 "CW" "" false [void Rig::setModeImpl(const QString &, const QString &, bool):../rig/Rig.cpp:433]
13:28:02.850 [DEBUG   ] [0x16bbcf000] [qlog.rig.rig.runtime]                             Waiting for Rig mutex [void Rig::setModeImpl(const QString &, const QString &, bool):../rig/Rig.cpp:435]
13:28:02.850 [DEBUG   ] [0x16bbcf000] [qlog.rig.rig.runtime]                             Using Rig [void Rig::setModeImpl(const QString &, const QString &, bool):../rig/Rig.cpp:435]
13:28:02.850 [DEBUG   ] [0x16bbcf000] [qlog.rig.driver.hamlibdrv.function.entered]       *** [virtual void HamlibRigDrv::setMode(const QString &, const QString &, bool):../rig/drivers/HamlibRigDrv.cpp:356]
13:28:02.850 [DEBUG   ] [0x16bbcf000] [qlog.rig.driver.hamlibdrv.function.parameters]    "CW" "" false [virtual void HamlibRigDrv::setMode(const QString &, const QString &, bool):../rig/drivers/HamlibRigDrv.cpp:358]
13:28:02.850 [DEBUG   ] [0x16bbcf000] [qlog.rig.driver.hamlibdrv.function.entered]       *** [virtual void HamlibRigDrv::setRawMode(const QString &):../rig/drivers/HamlibRigDrv.cpp:336]
13:28:02.850 [DEBUG   ] [0x16bbcf000] [qlog.rig.driver.hamlibdrv.function.parameters]    "CW" [virtual void HamlibRigDrv::setRawMode(const QString &):../rig/drivers/HamlibRigDrv.cpp:338]
```
- CW Confirmed by RIG - filter is too wide but it is OK 
```
13:28:03.109 [DEBUG   ] [0x16bbcf000] [qlog.rig.driver.hamlibdrv.function.entered]       *** [bool HamlibRigDrv::isRigRespOK(int, const QString, bool):../rig/drivers/HamlibRigDrv.cpp:972]
13:28:03.109 [DEBUG   ] [0x16bbcf000] [qlog.rig.driver.hamlibdrv.function.parameters]    0 "Get Mode Error" true [bool HamlibRigDrv::isRigRespOK(int, const QString, bool):../rig/drivers/HamlibRigDrv.cpp:974]
13:28:03.109 [DEBUG   ] [0x16bbcf000] [qlog.rig.driver.hamlibdrv.runtime]                Rig Mode:  2 Rig Filter:  2700 [bool HamlibRigDrv::checkModeChange():../rig/drivers/HamlibRigDrv.cpp:665]
[... cut ...]
13:28:09.985 [DEBUG   ] [0x16bbcf000] [qlog.rig.driver.hamlibdrv.runtime]                Object Freq:  "18.07310" [bool HamlibRigDrv::checkFreqChange():../rig/drivers/HamlibRigDrv.cpp:625]
```
- Rig repoted a filter change from 2100 -> 600Hz  - **WHY?**
```
13:28:10.510 [DEBUG   ] [0x16bbcf000] [qlog.rig.driver.hamlibdrv.runtime]                Object Mode:  2 Object Filter: 2100 [bool HamlibRigDrv::checkModeChange():../rig/drivers/HamlibRigDrv.cpp:666]
13:28:10.510 [DEBUG   ] [0x16bbcf000] [qlog.rig.driver.hamlibdrv.runtime]                emitting MODE changed "CW" "CW" "" 600 [bool HamlibRigDrv::checkModeChange():../rig/drivers/HamlibRigDrv.cpp:680]
```

foldynl, 2025-04-07:
Now QLog sends mode and then frequency. So should it be to send sequence freq and then mode?

a quick dirty patch without testing to change the order

```
diff --git a/ui/NewContactWidget.cpp b/ui/NewContactWidget.cpp
index 5fc8638..7427d6e 100644
--- a/ui/NewContactWidget.cpp
+++ b/ui/NewContactWidget.cpp
@@ -2748,6 +2748,10 @@ void NewContactWidget::tuneDx(const DxSpot &spot)
         return;
     }
 
+    frequency = ( frequency > 0.0 ) ? frequency : ui->freqRXEdit->value();
+
+    ui->freqRXEdit->setValue(frequency);
+
     if ( frequency > 0.0 )
     {
         QString subMode;
@@ -2782,12 +2786,7 @@ void NewContactWidget::tuneDx(const DxSpot &spot)
             emit userModeChanged(VFO1, QString(), mode, subMode, bandwidthFilter);
         }
     }
-    else
-    {
-        frequency = ui->freqRXEdit->value();
-    }
 
-    ui->freqRXEdit->setValue(frequency);
     resetContact();
     changeCallsignManually(spot.callsign, frequency);
 

```

aa5sh, 2025-04-07:
> foldynl

Quick test (I’m at work - shhhh) it did seem to be working correctly.  I will play with it some more tonight.

foldynl, 2025-04-07:
>>> Quick test (I’m at work - shhhh) 

me too

>>> it did seem to be working correctly.  

If that works, then it's pure magic.

jmeloranta, 2025-04-07:
I think wsjt-x has similar behavior. Have to sometimes set the band twice. I wonder if this because the radio remembers the mode that was used previously for the new band. So, setting the mode first does it for the current band and then changing the frequency to the new band picks up the mode from the radio that was last used there.

Jussi

foldynl, 2025-04-11:
@aa5sh  Did you have a chance to test it?

foldynl, 2025-04-11:
Many thanks for your testing.

foldynl, 2025-04-19:
@aa5sh, please, could you retest v0.43.1.  Maybe this change breaks ftdx-10 because I have a report from one user that everything worked for him in 0.43 and in 0.43.1 the Frequency is tuned incorrectly :
```
Hi!
May be it`s a bug? When i click on ssb station with frequensy 28.520in bandmap - my ftdx10 set to 28.519.350.
If i click again - setup correctly. This is only on SSB. In any band, but only first click. Any next click setup frequency correct.
```

foldynl, 2025-04-19:
This is a nightmare! We’ll probably need to approach it completely differently. As @jmeloranta mentioned, WSJTX does it like this – a comment from the commit.

```
commit 35e377d3649bfa14eca25d2f0ff07a0412c0154d
Author: Bill Somerville <g4wjs@classdesign.com>
Date:   Sat Feb 28 13:40:07 2015 +0000

    Use freq-mode-freq-mode sequency to set freqquency
    
    Because setting frequency can change  mode and setting mode can change
    frequency on various rigs it is  necessary to set a frequency by first
    setting a  frequency which  gets to the  correct band  and potentially
    band stack  register. Then  to set  the mode to  ensure that  any band
    remembered mode is cleared. Then set frequency again in case the prior
    mode change  moved the frequency  (e.g. CW correction). Then  set mode
    again because some rigs set the mode according to frequency.
    
    Who would have  thought setting the rig frequency via  CAT could be so
    complicated :( TODO: Even this is  flawed on some Yaesu rigs that must
    be band changed to select some settings like aerial socket and pre-amp
    settings.
    
    Merged from wsjtx-1.4 branch.
```
Why do the manufacturers do this, or why doesn’t Hamlib compensate for it!? It's just a rhetorical question.

aa5sh, 2025-04-20:
Sorry I’ve been out of town.  I’ll take a look at it this afternoon.  But I
think it may need to be something like the wsjtx solution
freq-mode-freq-mode sequence.  Which makes no sense like you say it
shouldn’t be necessary.  I’ll test it and get back with you.

foldynl, 2025-04-20:
> Sorry I’ve been out of town.  I’ll take a look at it this afternoon.  But I
> think it may need to be something like the wsjtx solution
> freq-mode-freq-mode sequence.  Which makes no sense like you say it
> shouldn’t be necessary.  I’ll test it and get back with you.
> […](#)

No rush. Commit [3e37ec9](https://github.com/foldynl/QLog/commit/3e37ec945427d5df628f3cc4544a776a093a4905) doesn’t work. Even though I call setFreq twice and call set mode again in case of a mode change, the FTDX10 behaves unpredictable. It started acting strange after the commit [c5d748a](https://github.com/foldynl/QLog/commit/c5d748a81dc7fb6c6c8e14b3a52c675a418ece66) - QLog v0.43.1 

I don’t know what to do anymore. Honestly, I’d rather revert version 0.43.1 commit [c5d748a](https://github.com/foldynl/QLog/commit/c5d748a81dc7fb6c6c8e14b3a52c675a418ece66) and the impacted operators will double click on the spot. Perhaps someone else can find a better approach.

aa5sh, 2025-04-20:
> … <x-msg://3/#>

Ok I was thinking this was a HamLib issue.  But I just connected my rig to my windows laptop and it did the same thing with being 700hz off when switching modes using OmniRig v1 as well.

aa5sh, 2025-04-20:
Ok there is a setting in the radio that is affecting this.  On my 101 it is:
FUNC -> CW SETTING -> MODE CW -> CW FREQ DISPLAY = PITCH OFFSET (or) DIRECT FREQ

There is a similar setting for the FTDX10 as well.  I changed it from PITCH OFFSET to DIRECT FREQ and the frequency seems to be working properly when clicking on spots.  I’m not sure what impact this has on the radio/operation if any.

foldynl, 2025-04-21:
Which change did you tested. You have 0.43.1 or the commit https://github.com/foldynl/QLog/commit/3e37ec945427d5df628f3cc4544a776a093a4905 with double call to setFreq?

aa5sh, 2025-04-21:
> foldynl
>  left a comment
> (foldynl/QLog#453)
> Which change did you tested. You have 0.43.1 or the commit 3e37ec9 with double call to setFreq?

With 0.43.1 it works correctly after this menu change in the radio.

foldynl, 2025-04-22:
perfect..may thanks for your tests. It means the commit https://github.com/foldynl/QLog/commit/3e37ec945427d5df628f3cc4544a776a093a4905 will not be necessary.

commit c5d748a8 (2025-04-19): Fixed #453 - Click on PHONE DX Spots sets wrong mode
closed 2025-04-22 by foldynl