foldynl/QLog #455: Wrong import of adif file of another log program

ik1vqy, 2024-09-06:
Good evening
I probably found an error in importing adif logs. 
When importing a log in adif format from another program, errors are always reported in the same serial numbers of the log itself, even if in reality the errors are not in the original log. Example:
[QSO#7001]: Error - A minimum set of fields are missing (start_time, call, band, mode, station_callsign) (2019-03-31; ; SSB)
[QSO#10001]: Error - A minimum set of fields are missing (start_time, call, band, mode, station_callsign) (2021-10-24; ; SSB)
[QSO#12001]: Error - A minimum set of fields are missing (start_time, call, band, mode, station_callsign) (2024-04-02; ; SSB)
If these are removed, in the new import of the correct log, the same reports are re-presented in the same serial numbers, even if the corresponding callsign has changed.
Furthermore, if callsigns are detected that do not correspond to DXCC countries, these are not imported into the QLog log. I think they should be imported anyway, but simply not counted in DXCC. In fact, it can happen that pre-existing DXCC entities no longer exist, but when they existed, the connections had been made and were valid.

Thanks and Best 73's

aa5sh, 2024-09-06:
I think I had similiar errors if I didn't have a Station Profile selected here.  Do you have one selected?  I could be wrong but I remember getting errors for every log like that and I believe it was the Station Profile being selected that corrected it:
[image]

ik1vqy, 2024-09-07:
Hi, and thanks for the answer.
I flag My Profile and also My Rig (sometimes... just to try...) but it doesn't work.
There is always the same error ad always in the same serial numbers : [QSO#7001], [QSO#10001], [QSO#12001], and sometimes also other numbers but always with the "001" as terminal. I don't know why...
regards

foldynl, 2024-09-08:
please, could you send us the impacted ADI records ?

ik1vqy, 2024-09-08:
Good Afternoon
Here are my adif Logs packed in zip format.
The "LogHX3-IK1VQY.adi" file is exported from LogHX3 program.
The "QLog-IK1VQY.adi" file is exported from QLog program and has not 14 records, the ones that the program has not imported signed as "errors" (see the image before).
Both the programs are running on Windows 11, but also QLog running on Linux (MXLinux 23.3) has the same issue.
Thanks
Best regards
Alfredo IK1VQY

[LogHX3-IK1VQY.zip](https://github.com/user-attachments/files/16922308/LogHX3-IK1VQY.zip)
[QLog-IK1VQY.zip](https://github.com/user-attachments/files/16922309/QLog-IK1VQY.zip)

aa5sh, 2024-09-08:
Odd I tried an import into a test db and the first time I got similar results as you above.  But I imported the file a second time it didn't error out on the last three entries.  I tried in a fresh test directory and I didn't get the last 3 errors.  I think those are generated because those lines don't have a STATION_CALLSIGN ADIF entry.  But there are other lines that don't have it so I'm not sure what is going on there..

As for the DXCC  - Ladislav - I wonder if someone doesn't have the Override DXCC checkbox selected should QLog try to lookup DXCC unless it is missing?  Maybe put a change similar to this in LogFormat.cpp. 

[image]

foldynl, 2024-09-09:
Please, could you send me a screenshot of the setting for your station profile "Home"?

foldynl, 2024-09-09:
please, add also a screenshot of the Help->About Dialog.

ik1vqy, 2024-09-09:
Hi Ladislav

Here are the screenshot of my settings. I added also the file that reports the errors.

Best regards

Alfredo

Best 73's de IK1VQY

>  
>  
> 
>  
> 
> Please, could you send me a screenshot of the setting for your station profile "Home"?

Riepilogo dell'importazione QLog

Data di importazione: 08/09/2024 22:19:29 UTC
File importato: C:/Users/ik1vq/Documents/IK1VQY-LOG/ik1vqy_20240909_001322.adi

Importati: 6 contatti
Allarmi: 12478
Errori: 14

Dettagli:
[QSO#1]: Attenzione - Duplicato (15/03/1995; VQ9XX; CW)
[QSO#2]: Attenzione - Duplicato (15/12/1995; IK2DUW; CW)
[QSO#3]: Attenzione - Duplicato (24/01/1998; SP7GIQ; CW)
[QSO#4]: Attenzione - Duplicato (27/02/2000; IW1REG; FM)
[QSO#5]: Attenzione - Duplicato (31/05/2000; I1EVB; FM)
[QSO#6]: Attenzione - Duplicato (04/12/2000; IW2MZK; FM)
[QSO#7]: Attenzione - Duplicato (04/12/2000; I7XNT; FM)
[QSO#8]: Attenzione - Duplicato (04/12/2000; IW2NDN; FM)
[QSO#9]: Attenzione - Duplicato (10/04/2001; I2KNJ; SSB)
[QSO#10]: Attenzione - Duplicato (10/04/2001; I2IVG; FM)
[QSO#11]: Attenzione - Duplicato (25/04/2001; I2UVZ; CW)
[QSO#12]: Attenzione - Duplicato (14/08/2001; I4STU; SSB)
[QSO#13]: Attenzione - Duplicato (14/08/2001; IW2NDW; SSB)
[QSO#14]: Attenzione - Duplicato (14/08/2001; IK2UJM; SSB)
[QSO#15]: Attenzione - Duplicato (09/03/2002; LW2EKD; SSB)
[QSO#16]: Attenzione - Duplicato (10/03/2002; I4ZFK; FM)
[QSO#17]: Attenzione - Duplicato (11/03/2002; IK5JWO; SSB)
[QSO#18]: Attenzione - Duplicato (11/03/2002; IZ2DOB; FM)
[QSO#19]: Attenzione - Duplicato (11/03/2002; IK2VTX; FM)
[QSO#20]: Attenzione - Duplicato (11/03/2002; IN3ZDC; SSB)
[QSO#21]: Attenzione - Duplicato (14/03/2002; IK1HZV; FM)
[QSO#22]: Attenzione - Duplicato (21/03/2002; 4Z5IW; SSB)
[QSO#23]: Attenzione - Duplicato (21/03/2002; F6IJU; CW)
[QSO#24]: Attenzione - Duplicato (23/03/2002; IZ2BKA; SSB)
[QSO#25]: Attenzione - Duplicato (24/03/2002; IZ2DBY; SSB)
[QSO#26]: Attenzione - Duplicato (18/04/2002; I0MYE; SSB)
[QSO#27]: Attenzione - Duplicato (18/04/2002; I0MYE; SSB)
[QSO#28]: Attenzione - Duplicato (19/04/2002; I1EVB; FM)
[QSO#29]: Attenzione - Duplicato (25/04/2002; IN3ZDC; SSB)
[QSO#30]: Attenzione - Duplicato (25/04/2002; I2RR; SSB)
[QSO#31]: Attenzione - Duplicato (05/05/2002; IW1AVF; SSB)
[QSO#32]: Attenzione - Duplicato (16/05/2002; IK1UGZ; FM)
[QSO#33]: Attenzione - Duplicato (16/05/2002; IK1UGZ; CW)
[QSO#34]: Attenzione - Duplicato (21/05/2002; IK1WJQ; CW)
[QSO#35]: Attenzione - Duplicato (21/05/2002; IK3HRP; SSB)
[QSO#36]: Attenzione - Duplicato (21/05/2002; I4HNB; SSB)
[QSO#37]: Attenzione - Duplicato (21/05/2002; IS0LDT; SSB)
[QSO#38]: Attenzione - Duplicato (22/05/2002; F2FS; CW)
[QSO#39]: Attenzione - Duplicato (25/05/2002; TA0/YU7AV; CW)
[QSO#40]: Attenzione - Duplicato (18/06/2002; SV8/IZ2CPP; SSB)
[QSO#41]: Attenzione - Duplicato (02/07/2002; DL6RO; CW)
[QSO#42]: Attenzione - Duplicato (03/07/2002; I1KTC; FM)
[QSO#43]: Attenzione - Duplicato (03/07/2002; IW1BWQ; FM)
[... 995 lines cut ...]
[QSO#1039]: Attenzione - Duplicato (26/06/2007; 7U5CI; SSB)
[QSO#1040]: Attenzione - Duplicato (26/06/2007; EA9HA; SSB)
[QSO#1041]: Attenzione - Duplicato (27/06/2007; IS0/4Z5KJ; SSB)
[QSO#1042]: Attenzione - Duplicato (02/07/2007; IE9/IK4RUX; SSB)
[QSO#1043]: Attenzione - Duplicato (03/07/2007; HI8LAM; SSB)
[QSO#1044]: Attenzione - Duplicato (07/07/2007; DL1BD; SSB)
[QSO#1045]: Attenzione - Duplicato (07/07/2007; LX/DC3SW/P; SSB)
[QSO#1046]: Attenzione - Duplicato (07/07/2007; DR2007SDW; SSB)
[QSO#1047]: Attenzione - Duplicato (08/07/2007; EI70FOY; SSB)
[QSO#1048]: Attenzione - Duplicato (08/07/2007; EA6CA; SSB)
[QSO#1049]: Attenzione - Duplicato (08/07/2007; IZ6IMN/P; SSB)
[QSO#1050]: Errore - Impossibile trovare le informazioni sull'entità DXCC (09/07/2007; 4N1GZ; SSB)
[QSO#1051]: Attenzione - Duplicato (09/07/2007; I5LN; SSB)
[QSO#1052]: Attenzione - Duplicato (09/07/2007; 4O6DJ; SSB)
[QSO#1053]: Attenzione - Duplicato (13/07/2007; 1A0KM; SSB)
[QSO#1054]: Attenzione - Duplicato (13/07/2007; CE4ETZ; SSB)
[QSO#1055]: Attenzione - Duplicato (16/07/2007; SU5HR; SSB)
[QSO#1056]: Attenzione - Duplicato (17/07/2007; BY2OMS; SSB)
[QSO#1057]: Attenzione - Duplicato (21/07/2007; OR4A; SSB)
[QSO#1058]: Attenzione - Duplicato (21/07/2007; J42POL; SSB)
[QSO#1059]: Attenzione - Duplicato (22/07/2007; II7GM; SSB)
[QSO#1060]: Attenzione - Duplicato (22/07/2007; ON3CYV/P; SSB)
[QSO#1061]: Attenzione - Duplicato (22/07/2007; YR1A; SSB)
[QSO#1062]: Attenzione - Duplicato (22/07/2007; IU3VMD; SSB)
[QSO#1063]: Attenzione - Duplicato (22/07/2007; IS

ik1vqy, 2024-09-09:
Good Afternoon
Here are my screenshots
[image]
[image]
[image]

Best regards

foldynl, 2024-09-09:
I am afraid that it will be necessary to clean the data in LogHX or attempt to export it as ADX. Here's what I found:

1) The LogHX export contains a large number of invalid entries, such as incorrect CQZ numbers (often 0, which is not defined), Mode is often marked as `????`, and other data inconsistencies. This is not as serious as the next point.

2) Some QTH entries contain non-ASCII characters what is not allowed in case of ADI file. Mixing UTF8 (2B charts) and ASCII (1B) chars causes unpredictable value - example HG7T's Name etc.

Personally, I don't fully understand the error yet - the issue is still under investigation. However, this input file is so damaged that even if QLog managed to load it, I wouldn't trust the data very much.

If possible, try generating an ADX file from LogHX (if that's an option) and try importing it to QLog. The ADX file supports non-ASCII characters. Maybe that will help the situation.

Regarding the old DXCC entities, yes, that would be nice-to-have, but it's not the main issue with the import.

foldynl, 2024-09-10:
It is opened a new issue for old DXCC Entity import issue - #459

ik1vqy, 2024-09-11:
HI
Many thanks for the work. I will try several ways to correct the log and I will try to import it.
I tried to correct the records that QLog said  to have not enough data direcly in LogHX and I exported only that records in an adif file. Then I have imported this file in QLog and the records were loaded correcly! OK! If instead I export the whole log in adif file the same errors re-appear...
Mistery!

Best regards