foldynl/QLog #467: Settings window extends beyond bottom of screen

KB3ITQ, 2024-09-21:
OS is Fedora 40.  The settings screen extends beyond the bottom of my screen, and I cannot access the buttons such as "cancel" and "save".  I installed Gnome Tweaks in the hopes it would add a maximize button.  It worked for the home logging page, but not for the settings window.  See attached screenshot.
[image]

Thanks!

foldynl, 2024-09-22:
What is your display resolution ?
Have you set the scale factor (screenshot below) to a value other than 100?

[image]

KB3ITQ, 2024-09-22:
My resolution is 1366 x 768.  I don't have the option to scale in the display settings (see screenshot).  I changed the scale from 1 to 0.5 in the monitors.xml file, and this hasn't appeared to have changed anything.
[image]

ik1vqy, 2024-09-23:
> OS is Fedora 40. The settings screen extends beyond the bottom of my screen, and I cannot access the buttons such as "cancel" and "save". I installed Gnome Tweaks in the hopes it would add a maximize button. It worked for the home logging page, but not for the settings window. See attached screenshot. ![Screenshot from 2024-09-21 15-48-15](https://private-user-images.githubusercontent.com/182372585/369644885-524b6931-44b2-45ec-8044-11f8ad1a00e2.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjcwOTM1MzUsIm5iZiI6MTcyNzA5MzIzNSwicGF0aCI6Ii8xODIzNzI1ODUvMzY5NjQ0ODg1LTUyNGI2OTMxLTQ0YjItNDVlYy04MDQ0LTExZjhhZDFhMDBlMi5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwOTIzJTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDkyM1QxMjA3MTVaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT04YmIwMjNjZmUwMGZmMTcwYWYzYjMxZmNiMjYyNmMzNmE0MTJhMjVjOWE5NGM3NmIxOWI1MDhmYTc0MWNjNzYwJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.rOOTTMQyY9emO3vltn-QBWkh1GVxyMZrZFzlDpZY91I)
> 
> Thanks!

Hi
You can push ALT and drag up the windows so You can see the save button...
Also in my laptop happens this...
No problem..

bg8ixz, 2024-09-23:
That's right, I've also found this kind of problem on screens with 1366 x 768 resolution.

KB3ITQ, 2024-09-24:
Thanks!  Alt - F7 works to move the window and then I can access the buttons below.

closed 2024-11-25 by foldynl