foldynl/QLog #473: Qlog won't connect to FT857+RigExpert Standart through Omnirig

Javier-EA6ALW, 2024-09-29:
Hi folks,

I am trying to connect my radio through Omnirig to Qlog but it doesn't work at all. However, radio is readed on all other softwares I have installed on my PC such as JTDX, Log4OM and so.

Configuration in Qlog seems quite simple and fast CAT is not working for it.

TNX in advance for your time.

Cheers!

Javier-EA6ALW, 2024-09-29:
Already found it! Newbie mistake on not checking on "connect to radio"

73!

Javier-EA6ALW, 2024-09-29:
Okay, although radio is connected, Qlog doesn't show frequency (stays at 0.00 Hz), but I'm able to switch mode (USB, LSB, CW....)

73!

foldynl, 2024-09-30:
Is the frequency option checked in the Rig settings?

Javier-EA6ALW, 2024-09-30:
Yes, it actually is. I found that setting the radio through hamlib works perfect (VFO, frequency, band, PTT and so). So maybe there's an issue on how Qlog reads OmniRig information with this radio.

Using hamlib prevents me of using JTDX, MSHV or other software simultanously.

(hope I make myself understand, English is not my first language and always try not to use translator)

TNX!
73

foldynl, 2024-09-30:
Do you use the omnirig v1 or v2 ?

Javier-EA6ALW, 2024-09-30:
> Do you use the omnirig v1 or v2 ?
v1

aa5sh, 2024-10-04:
Can you send a screenshot of your radio settings for OmniRig and in QLog?

Javier-EA6ALW, 2024-10-12:
> Can you send a screenshot of your radio settings for OmniRig and in QLog?

[image]

Javier-EA6ALW, 2024-10-12:
> Ok I just tested on a Windows machine with Smart SDR and it seemed to work ok with either OmniRig v1 or v2. Only issue I had was at first both were configured to use the same COM port and were fighting. Do you other applications work ok with Omnirig and your radio and all? I'd still be curious to see your setup for both settings in Qlog Hamlib and Omnirig as well as the Omnirig settings them self.

Sorry for the late response. Sure, I send you some pictures now

[image]

[image]

I hope everything is clear although it is in Spanish. TNX for your help

aa5sh, 2024-10-12:
So is OmniRig working with other applications?  Looks like there are some settings different between HamLib and Omnirig (Stop Bits) and Flow Control but that (Off versus High).

Javier-EA6ALW, 2024-10-13:
> So is OmniRig working with other applications? Looks like there are some settings different between HamLib and Omnirig (Stop Bits) and Flow Control but that (Off versus High).

Yes, it worked with everything. I tried to change stop-bits to the same number but nothing happened. I've been using LOG4OM and also used HDR back in the day.