foldynl/QLog #475: HamQTH dx Cluster error

lishengyou0521, 2024-09-30:
[image]
when i use qlog, always show this alert in the picture,it likes dx cluster has been closed,how to fix this error?

foldynl, 2024-09-30:
I'm sorry, but I don't understand what the dialog is showing. Can you translate it into English?

If you're having a problem with the HamQTH DXC connection, you can use others as well - see [Wiki DXC](https://github.com/foldynl/QLog/wiki/DX-Cluster)

lishengyou0521, 2024-09-30:
the dialog messages is cannot connected to dxc server,and the reason is host has been closed this connection

foldynl, 2024-09-30:
I would try to define some other, local DX Cluster.

lishengyou0521, 2024-09-30:
ok,i try it latter

closed 2024-11-04 by foldynl