foldynl/QLog #477: New function

BI6NSL, 2024-10-07:
[image]

Is it possible to add such functionality to the page?
To identify the frequency bands and modes in which the call sign has been operated and confirmed.
thank you

aa5sh, 2024-10-08:
I like this idea.  I think it would be nice to have something like this maybe:
[image]

foldynl, 2024-10-08:
Gentlemen,

Could you please provide a use case for this change? What is the reason or motivation for implementing it? The only reason I can think of is a DX Expedition. However, it seems to me that this table will be empty most of the time and, as a result, will take up unnecessary space in the GUI. Additionally, if you have the callsign, QLog already displays all QSOs for the given callsign. This allows the operators to see that they already have  QSOs with this station and check the status of those QSOs.

aa5sh, 2024-10-08:
Primary easy use case if for DX Expeditions like you say.  In my typically layout of QLog I only see about 2-3 previous QSOs depending on the device I use.  So I it is nice to quickly see a summary of previous QSOs.  But I also do work a lot of similar stations on FT8, FT4 on various bands.  That was one feature I missed from RumLogNG.  I do like the way you have the summary laid out better, I could never remember on RL which of the dots meant what.  It is not a have to have feature but it is interesting as I am working to see I’ve worked that station on 5 or more bands, without have to scroll up/down and look at bands/modes etc.

aa5sh, 2024-10-08:
Just for visual reference this is how I normally have it laid out.  And you can see this station I have had the pleasure to work 26 times.  
[image]

BI6NSL, 2024-10-08:
[image]

Although the list can scroll to see the Qsos that have worked, and the Qsos that have been confirmed, it is not graphically intuitive, and I often get confused about what frequencies and modes I have worked on the station, and which ones I have not worked on.
Avoid too many repetitive Qsos.
If you can hope that you can adopt the suggestion!
Thank you
DE BI6NSL Junxi

Down168, 2024-10-09:
> Just for visual reference this is how I normally have it laid out. And you can see this station I have had the pleasure to work 26 times. [...]

Although we know the number of passes , but we can't visualize the specific bands of this site's passes, the above two mentioned features, it is recommended to add to the new version, you can visualize the passes of the bands, modes, and the site's wave patterns, thank you.

foldynl, 2024-10-10:
To be honest, I don't think it should be included in the release at the moment. I'll take some time to think about it.

BI6NSL, 2024-10-12:
[image]

commit 46a01aec (2024-11-15): Fixed #477 - New Contact: Renamed DXCC Tab to DX Stats and contains DXCC and Station Statistics
commit 81d75055 (2024-11-16): Fixed #477 - QSO Detail: Renamed DXCC Tab to DX Stats and contains DXCC and Station Statistics
closed 2024-11-24 by foldynl