foldynl/QLog #479: Connection with Omnirig V2 to Xiegu G90 not working proper

thomasschwarz96, 2024-10-08:
Hello, I want to connect QLog v0.39.0 with Omnirig v2.1 to my Xiegu G90.
Operation System is Windows 10. 
I started Omnirig before QLog, as you mentioned in another Issue. 

I can change the mode as intended. 
But the Frequency is not updated, QLog shows `OOB` (see screenshots).

I tried several diffrent settings in Omnirig and QLog but I can`t get it working.
I also tried it with Omnirig v1, but there is the same issue. I also verified, that the COM-Port is the correct one.
Or did I anything wrong?

If you need further infos, just let me know.
The Rig works fine with N1MM+ or the Winlink Client (both on Windows 10).

I also attached Screenshots with my settings.
[image]
[image]
[image]

foldynl, 2024-10-08:
It looks that it is the same issue as #473

thomasschwarz96, 2024-10-08:
But with Hamlib I habe an similar problem. So I try it with Hamlib an Experiment with it

thomasschwarz96, 2024-10-08:
I got it working with hamlib. I used the settings of the Icom IC-7000, as shown in the screenshot below.
It is just a bit delayed, about 2-4 seconds I would say. Do you have for this case any tips?

[image]

foldynl, 2024-10-10:
>>> Do you have for this case any tips?

Besides the QLog poll interval, Hamlib also has own poll interval. Additionally, it may take some time to retrieve information from the radio. As I observed, there are many bugs in the G90 implementation in Hamlib. Unfortunately, due to Hamlib's release policy, I don't believe that these fixes will be included to the stable version in the near future I will have to think about what to do with this because Hamlib is not going to release a new stable release.

BI6NSL, 2024-10-13:
我用FLRIG和QLOG配合良好   并且让N1MM支持FLRIG的需求已经提交  
我现在使用的方式是FLRIG>WSJT>GT>QLOG
FLRIG用于控制钻机，
wsjt用于FT8
GT用于地图显示和通联数据的实时上传
qlog用于本地日志记录和管理以及警报

This method also applies to G90
=============================================
This method also applies to G90

I used FLRIG to work well with QLOG and the requirement for N1MM to support FLRIG has been submitted
The way I'm using it now is.
WSJT&gt;
GT&gt;
QLOG
FLRIG is used to control drilling RIGS,
wsjt is used for FT8
GT is used for map display and real-time upload of communication data
qlog is used for local logging and management as well as alerts

[image]

thomasschwarz96, 2024-10-14:
Just a quick Update: I found a perfect working setup!

I use `FLRig` with the Xiegu G90 Settings, see Screenshot below.
In `QLog` I use the FLRig Settings via Network, see Screenshot below. 

With this combination I got a instant / realtime change of the frequency, mode and so on.

[image]
[image]