foldynl/QLog #481: Add SWR monitor/alert from HamLib

vazquezrodjoel, 2024-10-13:
First of all, thanks for such a great piece of software!  Keep up the good work!

So I just installed today the latest version, so sorry if this is already in the works or considered as I am completely newby here.  I would suggest since you have the radio connection capability with Hamlib, would it be possible to monitor the SWR on the rig window?  

Here is my setup, I have hamlib running in another little computer and over the network I connect to it, passing audio and the hamlib connection.  On JTDX I can see that the hamlib provides the power and swr currently metered/sensed on by the radio on the top right corner (see image below.  Could that be implemented on QLog?  If so, going further, maybe an color alert would be great if say its higher than 2 or 3 (kinda green-yellow-red status).  
[image]

Just keep it in mind if possible! 
Thanks, Joel

foldynl, 2024-10-13:
The purpose of QLog is not to provide a rig control application. For rig control, there are specialized applications that display much more information about the rig. This is not the goal of QLog. The goal of QLog is to provide the operator with maximum information for logging and managing QSOs. In my opinion, SWR information is not relevant to this purpose. If you need to monitor the rig status remotely, there are applications like FLRig, wfview, and many others available for that.

BI6NSL, 2024-10-22:
Using flrig is a good choice, all data and state are clear
[image]

closed 2024-11-24 by foldynl