foldynl/QLog #482: Parse POTA park from the Cluster

vazquezrodjoel, 2024-10-14:
Could it be possible to parse the park from the cluster so that when you double click the contact, the POTA field be filled?

foldynl, 2024-10-14:
I would really like it. And not just POTA, but also SOTA, WWFF are here. However, the DXC message is free-text. So, the format is not very standardized, and it is entirely up to the operator what they write there. 

I have about 250k records from DXC for testing, and for example, POTA is reported in several ways. For example, K-0744 or K0344, sometimes as POTA and WWFF reference number follows !!!. There is even more chaos with WWFF. Sometimes as IFF0520, IFF-0520, or IFF 520. 

I’m not sure if I can capture all the variations of different notations. Do you know any reliable method for recognition? Just a note, searching for individual strings in a list of POTA/SOTA/WWFF is not possible. It is very time-consuming operation.

vazquezrodjoel, 2024-10-14:
I think the same way, if you try to cat them all it would be a nightmare, but I think at least as a start point it should parse those post that are in the correct format.  As a start point.  For example these 2 have them in the right format, clearly specified that are POTA.
DX de KD2OTI:    14344.0  K4RNY        [-POTA-] US-1241 SSB St. Vrain 2328Z
DX de KD7J:      14309.0  KA6BJR       [-POTA-] US-3553 SSB San Buena 2328Z

I think that should be a good start point.  Then you can add/modify to those stating POTA, but with no dash, like US1241, or case insensitive.  Thats as far as I would take it.  After this point, it would be a guess game and I think it would not benefit in the long run to either you or the end user.
I have no knowledge about SOTA or WWFF, I think is kinda similar in format the identification numbers, so I guess this concept could be applied/extended, but I would take it one at a time.  At least me... ;)

vazquezrodjoel, 2024-10-17:
Another idea that came to my mind now is that you can pull the pota active stations from the api (not sure the status on that, but assuming is working) and if a cluster entry does has word pota on the comment, then query the api to pull the park from that station.  But again, not sure if the api is working, google'ing I found this, but this is like too much for what you would need:

https://api.pota.app/spot/

If you think is might be possible/easy, I can try to find out the status of the api.  Let me know.

foldynl, 2024-10-19:
getting information via API is the slowest option there. But the positive news is that I did an analysis of the DXC messages and it looks like it can be implemented.

commit 848bd6ae (2024-12-16): Fixed #482 - Supressed POTA detection when IOTA string is present

IOTA and POTA have the same strings. So it's important to differentiate
them somehow. Sometimes people do it by adding IOTA. Try to use it.
commit f0739a93 (2025-01-11): Fixed #482 - DXC: Retrieve information about SOTA, POTA and WWFF
closed 2025-01-11 by foldynl