foldynl/QLog #484: Suggestion

BI6NSL, 2024-10-16:
[image]

It is recommended to make the contents of this box replicable, or export the txt to somewhere.
The user checks the cause of the mismatch one by one

aa5sh, 2024-10-17:
I was thinking the same thing this morning.  I had several fall into the unmatched section due to duplicate QSOs within the 30 minutes threshold.  But I couldn't copy and past this information so that I could then manually update those as being confirmed.  Had to do a screenshot like above.

foldynl, 2024-10-19:
The Upload and Download dialogs will change significantly. I will take your comments into consideration.

closed 2024-11-24 by foldynl