foldynl/QLog #485: fldigi to qlog  frequency  err

ik3ocd, 2024-10-20:
good evening ladislav, sorry for the disturbance I was observing this operation between fldigi and qlog it seems that fldigi frequency is not transferred correctly to qlog (see inserted images) for the rest it's fine.
fldigi ver 4.2.04, qlog ver .0.39 fedora linux 40. please verify if possible. 73 de ik3ocd flavio 
[image]

[image]

foldynl, 2024-10-22:
Unfortunately, I cannot replicate the issue with the same SW software. Please, could you run QLog in debug mode, perform the same logging actions as you show above and send me the output ?

How to run it in the debug mode:

`export QT_LOGGING_RULES=qlog.core.fldigi.*=true`
and then run QLog from command line.

foldynl, 2024-10-22:
I'm not sure if the frequency contains a comma or a period (your screenshot has a low resolution), but I was looking at GitHub for FLDIGI, and they were fixing something similar there. Could you please try the latest version (4.2.06) of FLDIGI?

By the way, the commit id is af361f7d6c38430a90b4694f7fc65dbccc6a994c and comment is
```
    ADIF specification compliance
        Change frequency field comma , to period .
```

ADIF records is imported with zero-freq by QLog if it contains comma.

foldynl, 2024-10-22:
Thanks. Unfortunately, it confirms my assumption. You have a version where freq is incorrectly sent. It is an FLDigi issue.

Your freq from FLDigi is:
```
<FREQ:8>7,040275
```

ADIF spec defines Freq as:

a sequence of one or more [Digits](https://www.adif.org/314/ADIF_314.htm#Digit) representing a decimal number, optionally preceded by a minus sign ([ASCII](https://www.adif.org/314/ADIF_314.htm#XL_ASCII) code 45) and optionally including a single decimal point  ([ASCII](https://www.adif.org/314/ADIF_314.htm#XL_ASCII)  code 46)

ik3ocd, 2024-10-22:
yes ladislav the frequency in on fldigi is not with the comma in the first separation and so:
7040,275
I hope the image resolution is good
I'll try a previous version of fldigi
[image]

ik3ocd, 2024-10-22:
ok thanks ladislav we await a more peaceful fldigi, thanks and see you soon.

closed 2025-04-05 by foldynl