foldynl/QLog #487: Hamlib 4.5.5 on Ubuntu 24.04

vazquezrodjoel, 2024-10-24:
Last week I upgrade my computer to Ubuntu 24.04 and also updates Hamlib to 4.5.5.  Qlog did not worked with radio.  I had to lower Hamlib to 4.5.4 and now is connecting fine.  

Before downgrading hamlib I grap a debug (hope I did it fine):

`QT_LOGGING_RULES=qlog.rig.*=true qlog
16:16:20.211 [DEBUG   ] [0x7235370eda40]                    [qlog.rig.rig.function.entered] => *** [Rig::Rig(QObject*):rig/Rig.cpp:24] 
16:16:20.212 [DEBUG   ] [0x72351aa006c0]                    [qlog.rig.rig.function.entered] => *** [void Rig::start():rig/Rig.cpp:164] 
16:16:20.869 [DEBUG   ] [0x7235370eda40]                    [qlog.rig.rig.function.entered] => *** [static qint32 Rig::getNormalBandwidth(const QString&, const QString&):rig/Rig.cpp:54] 
16:16:20.870 [DEBUG   ] [0x7235370eda40]                    [qlog.rig.rig.function.entered] => *** [void Rig::setFrequency(double):rig/Rig.cpp:390] 
16:16:20.870 [DEBUG   ] [0x7235370eda40]                 [qlog.rig.rig.function.parameters] => 0 [void Rig::setFrequency(double):rig/Rig.cpp:392] 
16:16:20.871 [DEBUG   ] [0x7235370eda40]                    [qlog.rig.rig.function.entered] => *** [void Rig::setMode(const QString&, const QString&):rig/Rig.cpp:444] 
16:16:20.871 [DEBUG   ] [0x7235370eda40]                 [qlog.rig.rig.function.parameters] => "AM" "" [void Rig::setMode(const QString&, const QString&):rig/Rig.cpp:446] 
16:16:20.871 [DEBUG   ] [0x72351aa006c0]                    [qlog.rig.rig.function.entered] => *** [void Rig::setModeImpl(const QString&, const QString&):rig/Rig.cpp:458] 
16:16:20.871 [DEBUG   ] [0x72351aa006c0]                 [qlog.rig.rig.function.parameters] => "AM" "" [void Rig::setModeImpl(const QString&, const QString&):rig/Rig.cpp:460] 
16:16:20.871 [DEBUG   ] [0x72351aa006c0]                             [qlog.rig.rig.runtime] => Waiting for Rig mutex [void Rig::setModeImpl(const QString&, const QString&):rig/Rig.cpp:462] 
16:16:20.872 [DEBUG   ] [0x72351aa006c0]                             [qlog.rig.rig.runtime] => Using Rig [void Rig::setModeImpl(const QString&, const QString&):rig/Rig.cpp:462] 
16:16:20.894 [DEBUG   ] [0x7235370eda40]                    [qlog.rig.rig.function.entered] => *** [void Rig::setMode(const QString&, const QString&):rig/Rig.cpp:444] 
16:16:20.894 [DEBUG   ] [0x7235370eda40]                 [qlog.rig.rig.function.parameters] => "CW" "" [void Rig::setMode(const QString&, const QString&):rig/Rig.cpp:446] 
16:16:20.894 [DEBUG   ] [0x72351aa006c0]                    [qlog.rig.rig.function.entered] => *** [void Rig::setModeImpl(const QString&, const QString&):rig/Rig.cpp:458] 
16:16:20.894 [DEBUG   ] [0x72351aa006c0]                 [qlog.rig.rig.function.parameters] => "CW" "" [void Rig::setModeImpl(const QString&, const QString&):rig/Rig.cpp:460] 
16:16:20.894 [DEBUG   ] [0x72351aa006c0]                             [qlog.rig.rig.runtime] => Waiting for Rig mutex [void Rig::setModeImpl(const QString&, const QString&):rig/Rig.cpp:462] 
16:16:20.894 [DEBUG   ] [0x72351aa006c0]                             [qlog.rig.rig.runtime] => Using Rig [void Rig::setModeImpl(const QString&, const QString&):rig/Rig.cpp:462] 
16:16:20.895 [DEBUG   ] [0x7235370eda40]                    [qlog.rig.rig.function.entered] => *** [void Rig::setMode(const QString&, const QString&):rig/Rig.cpp:444] 
16:16:20.895 [DEBUG   ] [0x7235370eda40]                 [qlog.rig.rig.function.parameters] => "CW" "" [void Rig::setMode(const QString&, const QString&):rig/Rig.cpp:446] 
16:16:20.895 [DEBUG   ] [0x72351aa006c0]                    [qlog.rig.rig.function.entered] => *** [void Rig::setModeImpl(const QString&, const QString&):rig/Rig.cpp:458] 
16:16:20.895 [DEBUG   ] [0x72351aa006c0]                 [qlog.rig.rig.function.parameters] => "CW" "" [void Rig::setModeImpl(const QString&, const QString&):rig/Rig.cpp:460] 
16:16:20.895 [DEBUG   ] [0x72351aa006c0]                             [qlog.rig.rig.runtime] => Waiting for Rig mutex [void Rig::setModeImpl(const QString&, const QString&):rig/Rig.cpp:462] 
16:16:20.895 [DEBUG   ] [0x72351aa006c0]                             [qlog.rig.rig.runtime] => Using Rig [void Rig::setModeImpl(const QString&, const QString&):rig/Rig.cpp:462] 
16:16:20.896 [DEBUG   ] [0x7235370eda40]                    [qlog.rig.rig.function.entered] => *** [static qint32 Rig::getNormalBandwidth(const QString&, const QString&):rig/Rig.cpp:54] 
16:16:20.896 [DEBUG   ] [0x7235370eda40]                    [qlog.rig.rig.function.entered] => *** [void Rig::setFrequency(double):rig/Rig.cpp:390] 
16:16:20.896 [DEBUG   ] [0x7235370eda40]                 [qlog.rig.rig.function.parameters] => 2.101e+07 [void Rig::setFrequency(double):rig/Rig.cpp:392] 
16:16:20.896 [DEBUG   ] [0x72351aa006c0]                    [qlog.rig.rig.function.entered] => *** [void Rig::setFrequencyImpl(double):rig/Rig.cpp:403] 
16:16:20.896 [DEBUG   ] [0x72351aa006c0]                 [qlog.rig.rig.function.parameters] => 2.101e+07 [void Rig::setFrequencyImpl(double):rig/Rig.cpp:405] 
16:16:20.896 [DEBUG   ] [0x72351aa006c0]                             [qlog.rig.rig.runtime] => Waiting for Rig mutex [void Rig::setFrequencyImpl(double):rig/Rig.cpp:407] 
16:16:20.896 [DEBUG   ] [0x72351aa006c0]                             [qlog.rig.rig.runtime] => Using Rig [void Rig::setFrequencyImpl(double):rig/Rig.cpp:407] 
16:16:21.162 [DEBUG   ] [0x72351aa006c0]                    [qlog.rig.rig.function.entered] => *** [void Rig::update():rig/Rig.cpp:173] 
16:16:21.162 [DEBUG   ] [0x72351aa006c0]                             [qlog.rig.rig.runtime] => Waiting for rig mutex [void Rig::update():rig/Rig.cpp:175] 
16:16:21.162 [DEBUG   ] [0x72351aa006c0]                             [qlog.rig.rig.runtime] => Updating Rig [void Rig::update():rig/Rig.cpp:182] 
16:16:21.295 [DEBUG   ] [0x7235370eda40]                    [qlog.rig.rig.function.entered] => *** [void Rig::setFrequency(double):rig/Rig.cpp:390] 
16:16:21.295 [DEBUG   ] [0x7235370eda40]                 [qlog.rig.rig.function.parameters] => 136000 [void Rig::setFrequency(double):rig/Rig.cpp:392] 
16:16:21.295 [DEBUG   ] [0x72351aa006c0]                    [qlog.rig.rig.function.entered] => *** [void Rig::setFrequencyImpl(double):rig/Rig.cpp:403] 
16:16:21.295 [DEBUG   ] [0x72351aa006c0]                 [qlog.rig.rig.function.parameters] => 136000 [void Rig::setFrequencyImpl(double):rig/Rig.cpp:405] 
16:16:21.295 [DEBUG   ] [0x72351aa006c0]                             [qlog.rig.rig.runtime] => Waiting for Rig mutex [void Rig::setFrequencyImpl(double):rig/Rig.cpp:407] 
16:16:21.295 [DEBUG   ] [0x72351aa006c0]                             [qlog.rig.rig.runtime] => Using Rig [void Rig::setFrequencyImpl(double):rig/Rig.cpp:407] 
16:16:21.295 [DEBUG   ] [0x7235370eda40]                    [qlog.rig.rig.function.entered] => *** [const QStringList Rig::getAvailableRawModes():rig/Rig.cpp:656] 
16:16:21.295 [DEBUG   ] [0x7235370eda40]                             [qlog.rig.rig.runtime] => Waiting for Rig mutex [const QStringList Rig::getAvailableRawModes():rig/Rig.cpp:658] 
16:16:21.295 [DEBUG   ] [0x7235370eda40]                             [qlog.rig.rig.runtime] => Using Rig [const QStringList Rig::getAvailableRawModes():rig/Rig.cpp:658] 
16:16:22.113 [DEBUG   ] [0x72351aa006c0]                    [qlog.rig.rig.function.entered] => *** [void Rig::update():rig/Rig.cpp:173] 
16:16:22.113 [DEBUG   ] [0x72351aa006c0]                             [qlog.rig.rig.runtime] => Waiting for rig mutex [void Rig::update():rig/Rig.cpp:175] 
16:16:22.113 [DEBUG   ] [0x72351aa006c0]                             [qlog.rig.rig.runtime] => Updating Rig [void Rig::update():rig/Rig.cpp:182] 
16:16:23.063 [DEBUG   ] [0x72351aa006c0]                    [qlog.rig.rig.function.entered] => *** [void Rig::update():rig/Rig.cpp:173] 
16:16:23.063 [DEBUG   ] [0x72351aa006c0]                             [qlog.rig.rig.runtime] => Waiting for rig mutex [void Rig::update():rig/Rig.cpp:175] 
16:16:23.063 [DEBUG   ] [0x72351aa006c0]                             [qlog.rig.rig.runtime] => Updating Rig [void Rig::update():rig/Rig.cpp:182] 
16:16:24.013 [DEBUG   ] [0x72351aa006c0]                    [qlog.rig.rig.function.entered] => *** [void Rig::update():rig/Rig.cpp:173] 
16:16:24.013 [DEBUG   ] [0x72351aa006c0]                             [qlog.rig.rig.runtime] => Waiting for rig mutex [void Rig::update():rig/Rig.cpp:175] 
[... cut ...]
16:16:28.003 [DEBUG   ] [0x72351aa006c0]                             [qlog.rig.rig.runtime] => Updating Rig [void Rig::update():rig/Rig.cpp:182] 
^C16:16:29.002 [DEBUG   ] [0x72351aa006c0]                    [qlog.rig.rig.function.entered] => *** [void Rig::update():rig/Rig.cpp:173] 
16:16:29.003 [DEBUG   ] [0x72351aa006c0]                             [qlog.rig.rig.runtime] => Waiting for rig mutex [void Rig::update():rig/Rig.cpp:175] 
16:16:29.003 [DEBUG   ] [0x72351aa006c0]                             [qlog.rig.rig.runtime] => Updating Rig [void Rig::update():rig/Rig.cpp:182] 
`

Hope that helps!

foldynl, 2024-10-25:
I can see just one issue - QLog is not able to connect to your RIGCTLD. Please, check if you are running rigctld with correct port and IP Address

```Network error 111: Connection refused\nnetwork_open: failed to connect to 127.0.0.1:4532```

vazquezrodjoel, 2024-10-25:
Yes, all was ok.  I checked all the settings, they were correct.  I created another rig with same settings, same result.  I tried JTDX, it worked fine.  I tried rigctl from terminal and it was connecting fine to the remote computer, but for some reason qlog never tried to reach that.  After couple of days I downgrade hamlib, and it just started working again when I got back to qlog.  Didnt even reboot the computer.

foldynl, 2024-10-25:
I have the same configuration, QLog is compiled from source, using hamlib 4.5.5 without any issue on Ubuntu 24.04. 

it could be caused by the fact that you have installed QLog from PPA for older Ubuntu. Check if you have the QLog PPA set to the "noble" (24.04) version after the system upgrade.

vazquezrodjoel, 2024-10-25:
Yes, just confirmed it has it for noble. 
`-rw-r--r-- 1 root root 1778 Oct 25 15:51 foldyna-ubuntu-qlog-noble.sources`

```
/etc/apt# apt policy qlog
qlog:
  Installed: 0.39.0-1~noble
  Candidate: 0.39.0-1~noble
  Version table:
 *** 0.39.0-1~noble 500
        500 https://ppa.launchpadcontent.net/foldyna/qlog/ubuntu noble/main amd64 Packages
        100 /var/lib/dpkg/status
```
        
I actually had to readd the cource and reinstall the package after the upgrade.

foldynl, 2024-10-28:
so what is the result, does it work for you now?

vazquezrodjoel, 2024-10-28:
No.  I had to downgrade hamlib to 4.5.4.  After downgrading it worked right away.

vazquezrodjoel, 2024-11-09:
Just to update, I dont know what it was, but yesterday I reinstalled the 4.5.5 version and it worked perfect this time... following the same installation directions... so nevermind. :)

closed 2024-11-09 by vazquezrodjoel