foldynl/QLog #492: tqsl/lotw

ik3ocd, 2024-11-02:
good morning ladislav, a new problem, since I updated fedora 40 to 41 there seems to be this problem with sending qso data to lotw I tried to update tqsl without positive results, maybe the new wayland graphical server that fedora uses? with the previous feora 40 and wayland tqsl had no problems, maybe op
[image]
enssl reading on error message, any positive ideas?. bye and thanks as always for help. 73 by ik3ocd flavio

foldynl, 2024-11-02:
Sorry, but this is not a QLog report, but a TQSL dialog window. QLog only uses the external TQSL program. If an error occurs in this external utility, it is necessary to contact the TQSL developer(s).

foldynl, 2024-11-02:
I just quickly found a similar problem in some other project. It may be a problem that the new Fedora disabled the SHA1 hash function. If that doesn't mean anything to you, it's something that's used to sign QSOs. Unfortunately, according to this guide https://lotw.arrl.org/lotw-help/developer-tq8/?lang=en TQSL uses this deprecated function when generating the QSO signature.

it is mentioned [here](https://github.com/dotnet/runtime/issues/65874) that `export OPENSSL_ENABLE_SHA1_SIGNATURES=1` can solve this. It must be defined in the user bash settings (for all applications).

foldynl, 2024-11-02:
And next description on FedoraWiki: https://fedoraproject.org/wiki/SHA1SignaturesGuidance

ik3ocd, 2024-11-04:
> flavio. It is not an Fedora issue. It is a big issue of TQSL. The use of
> SHA1 hash has long been marked as obsolete. Distributions have been warning
> a long time that this hash is outdated and will be disabled soon. And with
> the Fedora, that time has come.
>
> The important thing is that the workaround helped you.

Ok, thanks for the clarification ladislav ,yes I thought it was fedora,
lotw has been having some problems lately, but we're moving forward thanks
again and see you soon. regards

foldynl, 2025-04-05:
The TQSL team fixed this bug on their side a few months ago. I'm closing the issue.

closed 2025-04-05 by foldynl