foldynl/QLog #499: Add FCC db as Callsign info source.

vazquezrodjoel, 2024-11-11:
Is it possible to add the FCC db as a callsign information Source?

The DB information is hosted here: https://www.fcc.gov/uls/transactions/daily-weekly#weekly-files
In that page there is also a link for the information of the files.

I also found this script that will download the DB and import it to a MySQL DB, maybe can be used for sqlite?!
https://raw.githubusercontent.com/k3ng/hamdb/main/hamdb

foldynl, 2024-11-13:
I do not plan to add support for external callbooks (e.g., FCC, MMMONVHF.de) in the near future. It's essential to think this through carefully because there could be multiple offline external callbooks, each with its own format and set of information. How should this work alongside the current online callbook? There are still many questions. It’s not feasible for Qlog to include every possible external callbook published worldwide.

vazquezrodjoel, 2024-11-14:
Ok, I do see the point of the many offline DBs.  I just think it would be advantageous for the user to have a last resource backup for updated (free) information in case they do not subscribe to any of the other services.  For example I use HamQTH, but they don't always have the calls there.  I visualize it as I said, a last resource.  So for example same you have Primary and Secondary in the Query Order on the Call book settings, you could add a selection of some of those external logbooks in a drop-down selection box (hope that's the term) that I could select say FCC and the software downloads it, but only from official or governmental entities (meaning those that actually assign the call sign and privileges for that country).  Then same as you download your POTA, or SOTA files at startup, it could download in the background the weekly updates.  Just for the particular country of the station, otherwise would be a performance nightmare if it starts updated many services like those.  Just an idea.

foldynl, 2024-11-14:
Sorry, but I have a different opinion. Here are my arguments:

1) As I mentioned, there can be a large number of such databases, and unfortunately, each author has a different format. Maintaining any sort of callbook information is not a simple task.
2) Why should QLog specifically include the FCC (American database)? For people in Asia or Europe, a complete FCC database probably wouldn’t be very useful. I can already see people asking why, if I implement an offline callbook for one country, I don’t do it for another.
3) From my perspective, this point is probably the most important. QLog is primarily intended as logging software. It shouldn't implement functions provided by online services that are already in use and offer more detailed information. I understand your argument that QRZ is paid, but HamQTH is free. The fact that it doesn’t include all the information available in the FCC is a limitation of HamQTH and should be addressed on HamQTH's author(s).

vazquezrodjoel, 2024-11-15:
Make sense.  Thanks.

closed 2024-11-15 by vazquezrodjoel