foldynl/QLog #505: Testing 0.40 - Activities Menu

aa5sh, 2024-11-17:
I've been playing with it a little this afternoon.  So far I see a bunch of good stuff and changes have been made.  One thing I noticed was missing is the renamed 'Activities Menu' option.

[image]

aa5sh, 2024-11-17:
I was looking at the pull request.  I'm guessing I may need to run those sql commands to create the table?

aa5sh, 2024-11-17:
Ok after I ran it I still didn't notice it.  Then I just did where it is in the status bar at the bottom.

closed 2024-11-17 by aa5sh

foldynl, 2024-11-19:
I'm just writing the WIKI for this release. Yes, it's going to be a bit of a big change, but I thought it was better to move it to the status bar.