foldynl/QLog #510: Allow filter by Not-Worked (needed) band on Awards window

vazquezrodjoel, 2024-11-21:
It would be helpful if we could filter the needed bands on the award window.  Like in below, there are some states I need on some bands, but while I can select the "Not-Worked Only" filter, it will skip those bands since I have worked that entity before, but only on some bands.
[image]

With the "Not-Worked Only" filter:
[image]

aa5sh, 2024-11-22:
So if I understand you want to a selection for all bands or certain bands.  So you could pick 20,40,80 for instance and show which you haven't worked on those three bands?  Something like that?

vazquezrodjoel, 2024-11-22:
yes, for example in the second image if you notice lets say I want to focus to complete the WAS on 30m, 17m or 15m, it would be nice to be able to track what entities are needed.  Same with DXCC, etc...