foldynl/QLog #517: CW Macros might not working on CW for contests fields on v0.40.

vazquezrodjoel, 2024-11-24:
Had problems with the CW macros for contest exchange... was either not sending the exchange or sending the whole string like 
"5NN EXCHSTR TU"

Also, no the help when I hover the Profile fields, it says the field shold be EXCSTR, but the initial settings wiki says MYEXCHSTR, so maybe Im just entering the wrong shortcut...

foldynl, 2024-11-25:
Wiki has been fixed. As Michael mentioned, it should be `<EXCHSTR>` and it send your Exchange String.

just tested without any issue. My testing macro is:

`TEST <EXCHNR> <EXCHNRN> <EXCHSTR>`
the result is

`TEST 001 TT1 EXCHANGE`

with following QSO params:

[image]

vazquezrodjoel, 2024-11-25:
Ok, is not working.  Please notice in my screenshot, my settings, the field set on the new contact, and the cw console with just 5nn TU when I pressed F2.
[image]

Just FYI in case is relevand, I upgraded to v 0.40 over with apt-get upgrade on ubuntu 24.04.

vazquezrodjoel, 2024-11-25:
Ok, I find out that after making changes on the settings for the macros, I had to change the CW Profile on the CW Console.  It does not apply them right away.  Which is interesting, because when I set wrongly \<MYEXCHSTR\>, it did transmited that as a text, like 5NN MYEXCHSTR TU, so its weird...

So lesson learned, when making changes to the CW Profile, change the profile back and forward to apply the changes.

commit 98bf2206 (2024-11-29): Fixed #517 - CW Console: EXCSTR is not sent
closed 2024-11-29 by foldynl