foldynl/QLog #520: Request:  Allow to update TAB order no New Contact for Contest

vazquezrodjoel, 2024-11-25:
It would be more efficient if we can jump from Callsign field to Exchange field rather than the RST reports.  in contest with rst as part of the exchange everyone actually just sent 5nn for the most part (not that I agree with that, but the reality).  So I would like to jump from Callsign to the Exchanges (RcvNr or RcvExch).  But thats me, so maybe a way to customize the TAB order/flow would be great.

foldynl, 2024-11-26:
Use spacebar in this case. It is described on [Wiki](https://github.com/foldynl/QLog/wiki/Main-Window#new-qso):

`
QLog allows seamless navigation between individual fields within the main window by utilizing the tab bar. Additionally, a feature enhances the user experience: when the operator presses a Spacebar in the Callsign field, the application automatically shifts the focus to the first field following the RSTr field.
`

QLog goes even further. For all fields where a space is not expected, the spacebar is treated as a TAB

vazquezrodjoel, 2024-11-28:
Yep, that worked, thanks!

closed 2024-11-28 by vazquezrodjoel