foldynl/QLog #522: ClubLog upload problem

i0466lt, 2024-11-28:
Good evening,
I have a problem when uploading to ClubLog.
When I try to upload QSOs it returns an error window `Error transferring https://clublog.org/putlogs.php - server replied: ` with empty reply

Below I paste the info from the terminal window.

```
17:57:51.516 [INFO    ] [0x7f21c5251540]                    [qlog.ui.clublogdialog.runtime] => Clublog Upload Error:  "Operazione Clublog per il nominativo I2782VE non riuscita.<br>Error transferring https://clublog.org/putlogs.php - server replied: " [ClublogDialog::upload()::<lambda(const QString&)>:ui/ClublogDialog.cpp:142] 
```

I'm running QLog v.0.39.0 on Arch Linux.

Thanx,
Stefano SWL I0-466/LT

aa5sh, 2024-11-29:
Was it working previously with QLog and stopped or is this the first time you have tried to upload?  If it is the first time have you been able to upload successfully with another application?

i0466lt, 2024-11-29:
This is the first time I tried uploading with qlog. With other software it works normally. 
Tried with both username and password and username and application-password

aa5sh, 2024-11-30:
Out of curiosity since it looks like you are a SWL what do you have entered in the settings for Station Callsign and does that match your account at Clublog?  I'm not really familiar with how SWL's should upload to Clublog, but looks like it should work.  It looks like QLog is sending Email, Password and Callsign with upload to Clublog.  I think the callsign field is optional so it may need to be an exception for SWL's for that field to work.  Just very difficult to not have a useful error message from Clublog to help narrow it down.

aa5sh, 2024-12-02:
> SWL callsigns have the most absurd formats, over the years I've seen many
> of them. For example as *I0-466/LT* I use both in software and on clublog
> (but also eqsl, etc...) the callsign *I0466LT*
> Generally as a SWL I enter the callsign of the station heard in callsign
> field and the callsign of the station it was in QSO with in the comments
> field. Besides this clublog is useful for keeping track of worked/confirmed
> countries and as a backup of the log
> Is there a debug mode to actually see what it sends to ClubLog when you
> save the QSO?

Ok so you have the I0466LT entered in your station settings in QLog as
Station Callsign?

There is a debug output.  I have seen where Ladislav has directed people
how to enable that so he may can help you with that if I am directing you
wrong.

Here is the link to the wiki for debug level.  These commands are in Linux
format.  I know he directed someone the other day on how to do it for
Windows.
https://github.com/foldynl/QLog/wiki/Debug-Log-Level

So the command for Clublog debug I think would be something like this:
QT_LOGGING_RULES=qlog.core.clublog.*.debug=true ./qlog

i0466lt, 2024-12-02:
> Ok so you have the I0466LT entered in your station settings in QLog as
Station Callsign?

Yes correct. 
Thanks for the info on debugging, sorry but I really missed that part of the wiki. I've made a test with another callsign: I2782VE (already active on ClubLog), here is the output if it can be useful: (PASSWORD IS HIDDEN)

```
21:01:56.711 [DEBUG   ] [0x78215eee4540]               [qlog.core.clublog.function.entered] => *** [static const QString ClubLog::getEmail():core/ClubLog.cpp:58] 
21:01:56.711 [DEBUG   ] [0x78215eee4540]               [qlog.core.clublog.function.entered] => *** [static const QString ClubLog::getPassword():core/ClubLog.cpp:77] 
21:01:56.711 [DEBUG   ] [0x78215eee4540]               [qlog.core.clublog.function.entered] => *** [static const QString ClubLog::getEmail():core/ClubLog.cpp:58] 
21:01:56.713 [DEBUG   ] [0x78215eee4540]               [qlog.core.clublog.function.entered] => *** [static bool ClubLog::isUploadImmediatelyEnabled():core/ClubLog.cpp:68] 
21:02:01.416 [DEBUG   ] [0x78215eee4540]               [qlog.core.clublog.function.entered] => *** [static void ClubLog::saveUsernamePassword(const QString&, const QString&):core/ClubLog.cpp:85] 
21:02:01.416 [DEBUG   ] [0x78215eee4540]               [qlog.core.clublog.function.entered] => *** [static const QString ClubLog::getEmail():core/ClubLog.cpp:58] 
21:02:01.435 [DEBUG   ] [0x78215eee4540]               [qlog.core.clublog.function.entered] => *** [static void ClubLog::saveUploadImmediatelyConfig(bool):core/ClubLog.cpp:105] 
21:02:41.629 [DEBUG   ] [0x78215eee4540]               [qlog.core.clublog.function.entered] => *** [void ClubLog::insertQSOImmediately(const QSqlRecord&):core/ClubLog.cpp:371] 
21:02:41.629 [DEBUG   ] [0x78215eee4540]               [qlog.core.clublog.function.entered] => *** [const QString ClubLog::generateUploadCallsign(const QSqlRecord&) const:core/ClubLog.cpp:415] 
21:02:41.629 [DEBUG   ] [0x78215eee4540]               [qlog.core.clublog.function.entered] => *** [void ClubLog::sendRealtimeRequest(OnlineCommand, const QSqlRecord&, const QString&):core/ClubLog.cpp:116] 
21:02:41.629 [DEBUG   ] [0x78215eee4540]            [qlog.core.clublog.function.parameters] => 0 "I2782VE" [void ClubLog::sendRealtimeRequest(OnlineCommand, const QSqlRecord&, const QString&):core/ClubLog.cpp:120] 
21:02:41.629 [DEBUG   ] [0x78215eee4540]               [qlog.core.clublog.function.entered] => *** [static bool ClubLog::isUploadImmediatelyEnabled():core/ClubLog.cpp:68] 
21:02:41.629 [DEBUG   ] [0x78215eee4540]               [qlog.core.clublog.function.entered] => *** [static const QString ClubLog::getEmail():core/ClubLog.cpp:58] 
21:02:41.629 [DEBUG   ] [0x78215eee4540]               [qlog.core.clublog.function.entered] => *** [static const QString ClubLog::getPassword():core/ClubLog.cpp:77] 
21:02:41.629 [DEBUG   ] [0x78215eee4540]               [qlog.core.clublog.function.entered] => *** [static const QString ClubLog::getEmail():core/ClubLog.cpp:58] 
21:02:41.688 [DEBUG   ] [0x78215eee4540]                        [qlog.core.clublog.runtime] => "email=i2728ve%40gmail.com&callsign=I2782VE&password=HIDDENPASSWORD&api=21507885dece41ca049fec7fe02a813f2105aff2&adif=%3Ccall%3A5%3ELZ3GH %3Crst_sent%3A3%3E-15 %3Crst_rcvd%3A2%3E-1 %3Cfreq%3A5%3AN%3E7.074 %3Cband%3A3%3E40m %3Cmode%3A3%3EFT8 %3Cname%3A12%3EGeorgi Vlaev %3Cqth%3A11%3ENova Zagora %3Cdxcc%3A3%3E212 %3Ccountry%3A8%3EBulgaria %3Ccont%3A2%3EEU %3Ccqz%3A2%3E20 %3Cituz%3A2%3E28 %3Cpfx%3A3%3ELZ3 %3Cqsl_rcvd%3A1%3EN %3Cqsl_sent%3A1%3EQ %3Clotw_qsl_rcvd%3A1%3EN %3Clotw_qsl_sent%3A1%3EQ %3Ca_index%3A1%3E7 %3Cband_rx%3A3%3E40m %3Ccomment%3A6%3EPA3BBV %3Cemail%3A12%3Elz3gh%40dir.bg %3Ceqsl_qsl_rcvd%3A1%3EN %3Ceqsl_qsl_sent%3A1%3EQ %3Cfreq_rx%3A5%3E7.074 %3Ck_index%3A1%3E0 %3Cmy_antenna%3A8%3ELongwire %3Cmy_city%3A16%3ENoventa di Piave %3Cmy_country%3A5%3EItaly %3Cmy_cq_zone%3A2%3E15 %3Cmy_dxcc%3A3%3E248 %3Cmy_gridsquare%3A6%3EJN65GP %3Cmy_itu_zone%3A2%3E28 %3Cmy_name%3A7%3EStefano %3Csfi%3A3%3E186 %3Cstation_callsign%3A7%3EI2782VE %3Cqso_date%3A8%3AD%3E20241202 %3Ctime_on%3A6%3AT%3E200226 %3Cqso_date_off%3A8%3AD%3E20241202 %3Ctime_off%3A6%3AT%3E200241 %3Ceor%3E" [void ClubLog::sendRealtimeRequest(OnlineCommand, const QSqlRecord&, const QString&):core/ClubLog.cpp:169] 
21:02:41.915 [DEBUG   ] [0x78215eee4540]               [qlog.core.clublog.function.entered] => *** [void ClubLog::processReply(QNetworkReply*):core/ClubLog.cpp:266] 
21:02:41.915 [DEBUG   ] [0x78215eee4540]                        [qlog.core.clublog.runtime] => Clublog error URL  "https://clublog.org/realtime.php" [void ClubLog::processReply(QNetworkReply*):core/ClubLog.cpp:276] 
21:02:41.915 [DEBUG   ] [0x78215eee4540]                        [qlog.core.clublog.runtime] => Clublog error "Error transferring https://clublog.org/realtime.php - server replied: " [void ClubLog::processReply(QNetworkReply*):core/ClubLog.cpp:277] 
21:02:41.916 [INFO    ] [0x78215eee4540]                       [qlog.ui.mainwindow.runtime] => Clublog RT Upload Error:  "Operazione Clublog per il nominativo I2782VE non riuscita.<br>Error transferring https://clublog.org/realtime.php - server replied: " [MainWindow::MainWindow(QWidget*)::<lambda(const QString&)>:ui/MainWindow.cpp:271]
```

aa5sh, 2024-12-02:
Ok I don't see the error message that ClubLog is sending.  I think a line
may need to be added to the debug to show the response from ClubLog.  Based
on the notes in
https://clublog.freshdesk.com/support/solutions/articles/54906-how-to-upload-qsos-in-real-time
it says there should be an informative message returned.  In the code it
looks like it would just show network type errors, not messages returned in
the body.  Ladislav do you think somethink like adding this
reply->ReadAll() would show something more helpful as to what is happening?

[image: image.png]

Michael

aa5sh, 2024-12-02:
One thing I guess you could try to see the message is just in a web browser
copy and paste this url with your password put back in:
https://clublog.org/realtime.php?email=i2728ve%40gmail.com&
callsign=I2782VE&password=HIDDENPASSWORD&api=21507885dece41ca049fec7fe02a81
3f2105aff2&adif=%3Ccall%3A5%3ELZ3GH %3Crst_sent%3A3%3E-15
%3Crst_rcvd%3A2%3E-1 %3Cfreq%3A5%3AN%3E7.074 %3Cband%3A3%3E40m
%3Cmode%3A3%3EFT8 %3Cname%3A12%3EGeorgi Vlaev %3Cqth%3A11%3ENova Zagora
%3Cdxcc%3A3%3E212 %3Ccountry%3A8%3EBulgaria %3Ccont%3A2%3EEU
%3Ccqz%3A2%3E20 %3Cituz%3A2%3E28 %3Cpfx%3A3%3ELZ3 %3Cqsl_rcvd%3A1%3EN
%3Cqsl_sent%3A1%3EQ %3Clotw_qsl_rcvd%3A1%3EN %3Clotw_qsl_sent%3A1%3EQ
%3Ca_index%3A1%3E7 %3Cband_rx%3A3%3E40m %3Ccomment%3A6%3EPA3BBV
%3Cemail%3A12%3Elz3gh%40dir.bg %3Ceqsl_qsl_rcvd%3A1%3EN
%3Ceqsl_qsl_sent%3A1%3EQ %3Cfreq_rx%3A5%3E7.074 %3Ck_index%3A1%3E0
%3Cmy_antenna%3A8%3ELongwire %3Cmy_city%3A16%3ENoventa di Piave
%3Cmy_country%3A5%3EItaly %3Cmy_cq_zone%3A2%3E15 %3Cmy_dxcc%3A3%3E248
%3Cmy_gridsquare%3A6%3EJN65GP %3Cmy_itu_zone%3A2%3E28
%3Cmy_name%3A7%3EStefano %3Csfi%3A3%3E186 %3Cstation_callsign%3A7%3EI2782VE
%3Cqso_date%3A8%3AD%3E20241202 %3Ctime_on%3A6%3AT%3E200226
%3Cqso_date_off%3A8%3AD%3E20241202 %3Ctime_off%3A6%3AT%3E200241 %3Ceor%3E

i0466lt, 2024-12-03:
I can't test it through the browser because it has to be done through HTTP POST.

I tried it via CURL from the command line and it works:
```
curl -d "email=i2782ve@gmail.com&password=HIDDENPASSWORD&callsign=I2782VE&api=21507885dece41ca049fec7fe02a813f2105aff2&adif=%3Ccall%3A5%3ELZ3GH%3Crst_sent%3A3%3E-15%3Crst_rcvd%3A2%3E-1%3Cfreq%3A5%3AN%3E7.074%3Cband%3A3%3E40m%3Cmode%3A3%3EFT8%3Cname%3A12%3EGeorgiVlaev%3Cqth%3A11%3ENovaZagora%3Cdxcc%3A3%3E212%3Ccountry%3A8%3EBulgaria%3Ccont%3A2%3EEU%3Ccqz%3A2%3E20%3Cituz%3A2%3E28%3Cpfx%3A3%3ELZ3%3Cqsl_rcvd%3A1%3EN%3Cqsl_sent%3A1%3EQ%3Clotw_qsl_rcvd%3A1%3EN%3Clotw_qsl_sent%3A1%3EQ%3Ca_index%3A1%3E7%3Cband_rx%3A3%3E40m%3Ccomment%3A6%3EPA3BBV%3Cemail%3A12%3Elz3gh%40dir.bg%3Ceqsl_qsl_rcvd%3A1%3EN%3Ceqsl_qsl_sent%3A1%3EQ%3Cfreq_rx%3A5%3E7.074%3Ck_index%3A1%3E0%3Cmy_antenna%3A8%3ELongwire%3Cmy_city%3A16%3ENoventadiPiave%3Cmy_country%3A5%3EItaly%3Cmy_cq_zone%3A2%3E15%3Cmy_dxcc%3A3%3E248%3Cmy_gridsquare%3A6%3EJN65GP%3Cmy_itu_zone%3A2%3E28%3Cmy_name%3A7%3EStefano%3Csfi%3A3%3E186%3Cstation_callsign%3A7%3EI2782VE%3Cqso_date%3A8%3AD%3E20241202%3Ctime_on%3A6%3AT%3E200226%3Cqso_date_off%3A8%3AD%3E20241202%3Ctime_off%3A6%3AT%3E200241%3Ceor%3E" -H "Content-Type: application/x-www-form-urlencoded" -X POST https://clublog.org/realtime.php
```
Response: **OK**

If I resubmit the request I get **Dupe** back as expected.
For reference: [https://clublog.freshdesk.com/support/solutions/articles/54906-how-to-upload-qsos-in-real-time](https://clublog.freshdesk.com/support/solutions/articles/54906-how-to-upload-qsos-in-real-time)

foldynl, 2024-12-03:
Does it occur also in case when you upload the log via Service->Clublog? Based on your log, I can see that it is an error of the Realtime upload.

foldynl, 2024-12-03:
Oh, sorry. you wrote about it in the first comment.

foldynl, 2024-12-03:
Based on your comment above, you have 2FA enabled. please, could you confirm ?

i0466lt, 2024-12-03:
Hi Ladislav,
no, 2FA is disabled on my ClubLog account.

foldynl, 2024-12-03:
Can you please change your password to the simplest possible one and then try it? It's possible that you have a combination in your password that QLog will change in some way. Otherwise, I tried clublog upload with/without 2FA and everything works as it should.

i0466lt, 2024-12-03:
I changed the password to one consisting only of letters and numbers, the result doesn't change.
Tried uploading via POST from command line with CURL and it works.

foldynl, 2024-12-03:
I have retested all possible test cases for Clublog with my account and I don't see any issue. Unfortunately, the debug output said nothing about the issue. To be able to identify the issue I would need to get the HTTP Response code. Unfortunately, there is no debug message in the current QLog which is provides the information. 

you have two options:
1) If you are able to compile QLog from source code then apply this patch and test upload again

```
diff --git a/core/ClubLog.cpp b/core/ClubLog.cpp
index fc417a3..af936af 100644
--- a/core/ClubLog.cpp
+++ b/core/ClubLog.cpp
@@ -275,6 +275,7 @@ void ClubLog::processReply(QNetworkReply* reply)
     {
         qCDebug(runtime) << "Clublog error URL " << reply->request().url().toString();
         qCDebug(runtime) << "Clublog error" << reply->errorString();
+        qInfo() << replyStatusCode;
         if ( reply->error() != QNetworkReply::OperationCanceledError )
         {
             emit uploadError(tr("Clublog Operation for Callsign %1 failed.<br>%2").arg(reply->property("uploadCallsign").toString(),

```
2) if you are not able to compile QLog then you have to wait for the next release (approximately one month). I will add more verbose debug messages to Clublog and other external services to identify you issue.

i0466lt, 2024-12-03:
Ok thanks, at the moment I have no way to compile Qlog from source so I wait for the next release and try again.

I did some tests, installed from scratch Arch Linux on the notebook, tried with the Qlog package proposed by the system (yay) the problem is there too, tried with the latest flatpack release and everything works.
On the PC I use connected to the radios now I uninstalled everything and used the flatpack package and it works. At this point I think it is my problem related to something I had installed.

I thank you for your time, on the next release I will try again with the updated system package and let you know.