foldynl/QLog #532: IC-7300 CW Mode Constant Tone

jmcblane, 2024-12-01:
With QLog and my IC-7300, anytime I have the rig connected, if I switch to CW mode, it constantly sends out a tone. If I disconnect QLog, the rig stops transmitting. I have tried many settings to change this, but I can find nothing. There are no options for setting +/- DTR or +/- RTS, which I suspect may possibly be the cause?

aa5sh, 2024-12-01:
Maybe check these settings in your Radio for the USB port, if they are
turned on it may be causing this to happen I would think.
[image: image.png]

jmcblane, 2024-12-01:
You are correct. I have USB Keying for CW set to RTS. This works properly with other software, so that I can use the RTS line to key CW on the rig. However; with QLog, I'm not able to set the default value of the RTS line, and apprently it is by default set to key down constantly on this line.

For the time, I can set a profile to turn this off, but it would be nice to be able to control these values from QLog for the rig to have a proper workaround.

But, as not to appear ungrateful, which I am not, I very much appreciate QLog and recommend it to others. Thank you for your work! :)

aa5sh, 2024-12-01:
Maybe try to have flow control in QLog set to none and see if it works.  I
remember playing around with this with my 705 but don't remember well, it's
been a few weeks.

jmcblane, 2024-12-01:
Flow control is set to none when the problem happens.JACOB MCBLANE

ahit976, 2024-12-13:
> You are correct. I have USB Keying for CW set to RTS. This works properly with other software, so that I can use the RTS line to key CW on the rig. However; with QLog, I'm not able to set the [...]

I have the same problem with a DIY rig which protocol is compatible with the FT-817,as Qlog doesn't provide the RTS configuration now. I use COM3 connect with rig, and use COM4(USB to RS232) to virtualize handkey that is none working.

foldynl, 2024-12-13:
and doesn't it help if you switch PTT Type from CAT to RTS?

foldynl, 2026-02-09:
No reply, I'm closing this issue.

closed 2026-02-09 by foldynl