foldynl/QLog #533: error when upgrade from 0.38 to 0.40.1

TecnoTalarn, 2024-12-01:
Hi, I have a colleague who couldn't update ubuntu, and today I finally fixed his software sources.
The thing is that when updating Qlog it stopped working
With the following error:

qlog: symbol lookup error: qlog: undefined symbol: rigerror2

He had the version: 
qlog_0.38.0-1~jammy_amd64.deb
and when installing:
qlog_0.40.1-1~noble_amd64.deb
generates the error.
I left it with version 0.38 and everything works again.

Surely it must be missing some software library? Or can it be because of ubuntu version? jammy or noble?

Thank you!

TecnoTalarn, 2024-12-01:
Yes, he is using 24.04.1 LTS with PPA.
He may have an old version of hamlib, when I can I will connect to it and check. But it sounds to me that he does not have 4.5.5
The easiest thing will be to uninstall qlog and hamlib and try to install the latest qlog and see what happens.
He does not use the communication with the radio, since it is an old ts440 without the communication chip, therefore, in principle hamlib does not use it at all.

many thanks!

foldynl, 2025-05-09:
I guess it hasn't appeared. I am closing the issue

closed 2025-05-09 by foldynl