foldynl/QLog #535: Request:  A computer migration tool/option

vazquezrodjoel, 2024-12-02:
It would be useful a migration option where you can export your settings from computer to computer, like from Linux to Windows or Mac so users don't have to go over the whole configuration over again.  Not the log, but the settings, usernames, passwords, etc.  

Passwords can be tricky to keep them encrypted... maybe...  but just consider the possibility.

foldynl, 2024-12-02:
I would like that. Unfortunately, I made a few mistakes during development and it's not that easy to implement it now. But I am working on fixing it. Even so, there are a few things that currently prevent migration between platforms. Passwords migration is only one of them. Fortunately, there is an export QSOs. I don't think that migrating between platforms would be that common operation. I would rather see a problem when upgrading computers, but that is solved by backing up the Home directory.

vazquezrodjoel, 2024-12-02:
Yeah, I just had the situation last night where I got a new computer and (against my main instinct) will give Windows in my shack an opportunity and got Qlog setup, but going over all the configuration I thought some export would be definitely needed.  And if the backups can save that configuration export that would be even better, so once you have the software setup you can do an export and backup the settings.  We can setup windows after, but at least the main setup would be there.  

Maybe not all initially, but everything possible from the Settings window(s).

Like always, please let me know if you need me to test that I can help with.  This is a great software and I love it so far as many other operators.  Thanks for everything you and the other developers do on it.

BI6NSL, 2025-01-19:
Great advice