foldynl/QLog #536: New ADIF 3.1.5 Spec

foldynl, 2024-12-02:
Full changelog [here](https://www.adif.org/315/ADIF_315_annotated.htm).

The following points have an impact on QLog:
- Addition of QSO Fields: [CNTY_ALT](https://www.adif.org/315/ADIF_315_annotated.htm#QSO_Field_CNTY_ALT), [DCL_QSLRDATE](https://www.adif.org/315/ADIF_315_annotated.htm#QSO_Field_DCL_QSLRDATE), [DCL_QSLSDATE](https://www.adif.org/315/ADIF_315_annotated.htm#QSO_Field_DCL_QSLSDATE), [DCL_QSL_RCVD](https://www.adif.org/315/ADIF_315_annotated.htm#QSO_Field_DCL_QSL_RCVD), [DCL_QSL_SENT](https://www.adif.org/315/ADIF_315_annotated.htm#QSO_Field_DCL_QSL_SENT), [MORSE_KEY_INFO](https://www.adif.org/315/ADIF_315_annotated.htm#QSO_Field_MORSE_KEY_INFO), [MORSE_KEY_TYPE](https://www.adif.org/315/ADIF_315_annotated.htm#QSO_Field_MORSE_KEY_TYPE), [MY_CNTY_ALT](https://www.adif.org/315/ADIF_315_annotated.htm#QSO_Field_MY_CNTY_ALT), [MY_DARC_DOK](https://www.adif.org/315/ADIF_315_annotated.htm#QSO_Field_MY_DARC_DOK), [MY_MORSE_KEY_INFO](https://www.adif.org/315/ADIF_315_annotated.htm#QSO_Field_MY_MORSE_KEY_INFO), [MY_MORSE_KEY_TYPE](https://www.adif.org/315/ADIF_315_annotated.htm#QSO_Field_MY_MORSE_KEY_TYPE), [QRZCOM_QSO_DOWNLOAD_DATE](https://www.adif.org/315/ADIF_315_annotated.htm#QSO_Field_QRZCOM_QSO_DOWNLOAD_DATE), [QRZCOM_QSO_DOWNLOAD_STATUS](https://www.adif.org/315/ADIF_315_annotated.htm#QSO_Field_QRZCOM_QSO_DOWNLOAD_STATUS), [QSLMSG_RCVD](https://www.adif.org/315/ADIF_315_annotated.htm#QSO_Field_QSLMSG_RCVD).
- Changes to the [Contest ID](https://www.adif.org/315/ADIF_315_annotated.htm#Contest_ID) enumeration:

    Correction to the CVA-DX-SSB contest description, which specified CW instead of SSB.
    Addition of new Contest IDs:
    "ARI Italian Activity Contest (VHF)", "ARI Italian Activity Contest (UHF)", "ARI Italian Activity Contest (6m)", "ARI Italian Activity Contest (23cm)", "ARI Italian Activity Contest (13cm+)", "ARI Italian EME Trophy", "DARC FT4 Contest", "K1USN Slow Speed Open", "PCC Pro CW Contest".

- Addition to the [Mode](https://www.adif.org/315/ADIF_315_annotated.htm#Mode_Enumeration) and [Submode](https://www.adif.org/315/ADIF_315_annotated.htm#Submode_Enumeration) enumerations of submodes FSKH245 and FSKH105.
- Changes to the [VUCC_GRIDS](https://www.adif.org/315/ADIF_315_annotated.htm#QSO_Field_VUCC_GRIDS) and [MY_VUCC_GRIDS](https://www.adif.org/315/ADIF_315_annotated.htm#QSO_Field_MY_VUCC_GRIDS) fields' descriptions to specify six as well as four character locators to bring them into line with the LoTW implementation.

aa5sh, 2024-12-02:
That's kind of interesting for them to put the morse code key info in the
ADIF spec.  Wonder why they were adding that they didn't also do
microphones.

foldynl, 2024-12-02:
Try the ADIF mailing list; it has been discussed there quite a lot. What bothers me more is that ADIF keeps expanding with new fields that only a few people actually use. It's impossible for an operator to fill in every fields in the ADIF. 

The approach to online services stirs strong emotions in me because the authors seem to be heading toward each service having its own field. In a time when almost every country runs its online callbook, this is a pretty poor approach. I need to stay calm about it.

aa5sh, 2024-12-02:
I was aligning with your train of thought in that I don't understand adding
4 fields for CW keys, much the same as the QRZ info.  That should fall into
custom fields or something, like you said there are a bunch of log services
and it shouldn't account for them all.  There has to be a better way.

aa5sh, 2024-12-02:
I like the way you just have the fields entry.  For instance I have an
email that I have used in the past to email out QSL cards for new stations
I worked.  I just need a place to stuff a I did it flag.  But don't need it
for anything else than me.  I haven't taken the time ot rewrite my app to
look at QLog database but it should be trivial to grab the data and update
that field so it is tracked.

foldynl, 2025-01-11:
v0.41 includes the PR

closed 2025-01-11 by foldynl