foldynl/QLog #538: 60m band now showing in DX stats & display total # of slots in awards

jmeloranta, 2024-12-02:
1) I have enabled 60m band in the settings but when looking at DX stats after entering a call sign, it does not show 60m band data. 

2) Logbook -> Awards could include one more column that shows the total number of slots (= all bands combined; sum of 160m through 6m). For example, clublog shows this data.

foldynl, 2024-12-03:
>>> I have enabled 60m band in the settings but when looking at DX stats after entering a call sign, it does not show 60m band data.

DX Stats displays a table based on [DXCC rules](http://www.arrl.org/dxcc-rules), where the 60m band is not defined. It is possible to enable the 60m band for Station Statistics, but this results in both tables having a different number of columns, which may be user-unfriendly. Therefore, we have two options:

1) Expand both tables to include all active bands, which will deviate from DXCC rules, requiring us to rename the table to DXCC Entity Statistics.
2) Leave the tables as they are, excluding the 60m band.

>>> Logbook -> Awards could include one more column that shows the total number of slots (= all bands combined; sum of 160m through 6m). For example, clublog shows this data.

It's worth considering.

jmeloranta, 2024-12-04:
Regarding 60m band, I think it would be logical to show bands there that are enabled?

kyleboyle, 2024-12-04:
> Regarding 60m band, I think it would be logical to show bands there that are enabled?

The data in the DXCC table is meant to show your progress towards DXCC accreditation within the context of the currently logged QSO. the 60m band isn't part of that accreditation system.

jmeloranta, 2024-12-04:
People do different things. I am collecting band counters. The best solution would to show those bands that are enabled. If you are not interested in 60m band, disable it in the settings.

commit d6f16c74 (2024-12-20): Fixed #538 - DX Stats contain all enabled bands - not only DXCC Bands
commit 5af72497 (2025-02-14): Fixed #538 - Added Slots - total over each band
closed 2025-01-11 by foldynl