foldynl/QLog #539: DXC: DIGITAL spots should tune USB

foldynl, 2024-12-03:


vazquezrodjoel, 2024-12-03:
Maybe provide an option to other modes specific to the radio?!  Like I use my 991a on Data mode...  Just want to mention it as a heads-up, because this could affect other users setups.

foldynl, 2024-12-03:
:-( Next option ? Unacceptable. It needs to be considered whether the additional information can be obtained in another way. It might work here.

vazquezrodjoel, 2024-12-03:
The thing is that there are a few variables that can affect this.  How is presented on the DXC, what are the radio capabilities, what are the user preference with that particular radio.  Thats only the ones I can think of real quick here.  I would make a drop down selection (string value) on the Rig profile on what mode the user wants to push to the radio when a "DIGI" mode contact is selected on the cluster.  and provide only USB, DATA.  I mean not all the modes, but just the phew ones that are related to digital modes.  I don't think it can be pulled from the radio model specific because for example I use Hamlib net rigctl (I think thats it, Im not in front of the shack computer now), and that "rig" can apply to many...

Just picking your brain ;)

BI6NSL, 2024-12-05:
That's a great offer. I've been stuck on that for a long time

foldynl, 2024-12-08:
From qlog mailing group:

here is what is requested:
Radio Flex 6400 in DIGU mode
JTDX in FT8 mode
Qlog open in SSB/USB

Active Qlog radio card with the following parameters:
Hamlib interface, FlexRadio 6xxx (20220306.0)
flags on: Mode, VFO, Frequency, clear QSY, CW keyer speed, sync CW speed
Polling 100 ms
Port type: network [127.0.0.1:7821](http://127.0.0.1:7821/)
 
DX cluster configured on Hamqth.com:7300 (for now)
cluster filters mode: CW, Digital, FT8

test 1-
spot cluster FT8 mode frequency 28.074 - double click
result: Frequency ok on radio, jtdx and qlog
callsign ok on qlog
mode on  radio:USB
mode on qlog: SSB/USB

test 2-
spot cluster CW mode frequency 14.151 - double click
result: Frequency ok on radio, qlog
callsign ok on qlog
mode on radio: CW
mode on qlog: CW

test 3-
spot cluster mode DIGITAL frequency 7.0470 - double click
result: Frequency ok on radio, (JTDX) and qlog
callsign ok on qlog
mode on radio: LSB
mode on qlog: SSB/LSB

foldynl, 2024-12-08:
I tested and according to the source code, SSB/USB is always sent to the rig when the mode is DIGITAL and FT8. My question is. Is it possible that you have some automatic LSB/USB selection enabled on the rig?

foldynl, 2024-12-08:
Ladislav good afternoon,
sorry for the long delay, but I was waiting for a hermesLite 2 and I wanted to answer you after installing and also trying with the one that uses TCI as a protocol to connect the radio.
Both trx behave exactly the same way, with the only difference that:
Flex6400 when I set DIGIU in the SmartSDR program of the radio, in the Qlog Radio window it appears as a transmission mode PKTUSB
Hermes lite 2 when I set DIGIU in the Thetis v2.10.3.7 beta1 program, in the Qlog Radio window it appears DIGU
I don't think I have "configurations" in the two radios that select LSB/USB autonomously.
If you need other tests I'm available.

73 Roberto IU2NUB

foldynl, 2025-01-11:
v0.41 includes some changes related to this issue

closed 2025-01-11 by foldynl