foldynl/QLog #549: Web Lookup Button

dl2ki, 2024-12-14:
**System: Kubuntu 24.04.1 LTS, QLog 0.40.1**

Hi,

I noticed today that the “Web Lookup Button” in the main screen apparently has no function here.
 The Web Lookup Button option in the settings menu is `https://www.qrz.com/lookup/<DXCALL>`.

foldynl, 2024-12-14:
Wolfgang, QLog call the default browser to open the URL. I have tested under Kubuntu 24.04.1 the the same QLog version that the QLog can open the URL. Please, could you verify, whether you have selected the default browser in Kubuntu System Setting -> Default Applications -> Web browser. Then try to open terminal, type www.google.com, right-click mouse button and select Open Link. Will it open the browser ?

dl2ki, 2024-12-14:
Hi Ladislav,

Thanks for the tip. Obviously my mistake.

I had already checked the settings in System Preferences -> Default Applications -> Web Browser. It said 'Firefox' as the default browser.

[image]

After opening the drop down menu, I found another Firefox entry there.
After selecting it, the 'Web Lookup Button' now works again.

[image]

Apparently this old entry was left over after the SNAP installation of Firefox was replaced by the Mozilla PPA after the Kubuntu upgrade. So far this has not been noticed here. I will have to pay attention to that in the future.

Sometimes it needs an external hint, hi. Thanks again.

The issue can be closed.

closed 2024-12-15 by foldynl