foldynl/QLog #550: Function request: Add the matching QSO communication information label printing function, similar to the following effect picture

BI6NSL, 2024-12-14:
Function request: Add the matching QSO communication information label printing function, similar to the following effect picture

[image]
[image]

BI6NSL, 2024-12-14:
I sincerely hope that you can consider this request.

foldynl, 2024-12-14:
This has already been discussed. Implementing a graphical editor in QLog seems unnecessary to me because there are many applications that do it much better. For example, using CSV export and [gLabels](https://help.gnome.org/users/glabels/stable/) can create exactly what you need.

BI6NSL, 2024-12-15:
gLabels do not interact with Qlog QSO data

aa5sh, 2024-12-15:
I think he is referring to you could export a CSV file then import in into gLabels to merge and print after creating a suitable template.

aa5sh, 2024-12-15:
Like explained [here](https://help.gnome.org/users/glabels/stable/merge-ex2.html.en).

BI6NSL, 2024-12-15:
> 我认为他指的是您可以导出 CSV 文件，然后在创建合适的模板后将其导入 gLabels 进行合并和打印。

It should be, but I can't do it yet

closed 2025-01-19 by BI6NSL