foldynl/QLog #553: search of short callsign

TecnoTalarn, 2024-12-19:
Hello, I have detected a problem with the correspondent search
IF the callsign is short, for example 7Q1, it does not find it in the list if I make a new contact.
It seems to expect longer callsigns, 4 characters or more... In normal conditions it is sufficient, but in contests or DXpeditions they may have shorter callsigns.
[image]

If I search for the contact in the database, it does appear.

[image]

aa5sh, 2024-12-19:
That's weird I just entered K2P for a station I worked yesterday and it did seem to work.  It looks like the code is written to be 3 or more characters.
[image]

foldynl, 2024-12-19:
As @aa5sh mentioned, QLog searches for callsigns that are three or more characters long. It's important to note that it searches for valid callsigns. Unfortunately, I don't know which authority issued this callsign, but it is not a valid callsign—it's only a prefix. The suffix is missing. Since it's not a valid callsign, QLog skips the search.

If there will be more such "anomalies", I will try to change it. At this moment I leave it unchanged.

foldynl, 2024-12-19:
in this case, `1` is number separates prefix and suffix. The suffix is defined as character string (source: https://en.wikipedia.org/wiki/Amateur_radio_call_signs). If you have batter source, please, could you share it ?

kyleboyle, 2024-12-19:
I see what you mean now. The heuristic referenced for operator callsigns doesn't apply to special event callsigns which can be '1x1 callsigns'. I guess it's a question of whether qlog should handle these special event calls correctly or not

foldynl, 2024-12-19:
If more such anomalies (invalid callsigns) occur in the future, I will fix them.

Unfortunately, it's not just a matter of fixing one regex for callsigns, but a revision of the QLog, as I'm not sure where else it might cause problems.

TecnoTalarn, 2024-12-20:
On this topic
with a longer callsign it also happens, when I send it from WSjtx, it does not search, although in the station statistics the callsign and the bands worked do appear.
If I enter it manually it appears correctly at the top.

[image]