foldynl/QLog #555: Frequency input problem in v0.40.1

DL2XT, 2024-12-19:
After the update to the latest version `0.40.1` I have a problem with the frequency input field. 

If I put in e.g. the frequency `431.900` or `431.9` I got an error `submm` or `OOB`.
The application converts the frequency incorrect. It converts it to `43190000000`.

I only updated QLog and after it I got the problem if I enter manually the frequency.
This is also the case, if I want to edit an QSO.
Before it, it worked perfect. 

Below you see some screenshots:

[image]
[image]
[image]

DL2XT, 2024-12-19:
I tried this as well @ik1vqy.
But I can't enter an comma. This will not be entered, just a dot.

aa5sh, 2024-12-19:
Can you post a copy of the error message?  If I try that without a radio connected it works as expected so I wonder if it rig control that is throwing the error or handling it improperly. I don't have a cable hooked up to my VHF/UHF rig now to try I can this evening.

DL2XT, 2024-12-19:
@aa5sh There is now radio connected. I just start QLog and then enter, as described above, the frequency.
So there is no 3rd party tool in place which could cause that error.

aa5sh, 2024-12-19:
I agree with @ik1vqy that this seems like a localization thing with numbers. What is your computer locale set to?  I would think you should be able to enter 431,9 and it would work ok.  What you are seeing I get when I enter 431,900 or 431,9 but works ok for me as 431.900

DL2XT, 2024-12-19:
So in the meantime I tried to enter the numbers with an comma. So I copied the value `431,9` or `431,900` or even `145,500` but it is always converted wrong. I made no changes regarding the language or similar. I just upgraded to `v0.40.1` and now I got this error. Before it worked like a charm.

DL2XT, 2024-12-19:
I also  uninstalled `QLog` and then installed it again but the issue still remains

aa5sh, 2024-12-19:
I think you said above that you are not using Rig control.  Do you have any
Rigs defined and not used?  I'm just trying to see if I can find a way to
duplicate what you are seeing.  Doesn't really make sense what is happening.

DL2XT, 2024-12-19:
I had different rigs defined, but I also deleted those, but no change. 
I don't know what is happening here.

foldynl, 2024-12-19:
Linux or Windows ?

DL2XT, 2024-12-19:
Windows, sorry I forgot that!

foldynl, 2024-12-19:
run power-shell and execute follow command
```powershell -Command "Get-Culture | Select-Object -ExpandProperty NumberFormat | Select-Object -ExpandProperty NumberDecimalSeparator"```

please, could you send the result ?

foldynl, 2024-12-19:
Your decimal separator is a period. 

I tested it under Win10 and Win11 and the behavior is the same. If I try to insert something other than the correct decimal separator, it is not inserted. Unlike Linux, where it can be inserted. However, if I insert the correct separator, in my case comma, and press enter, the frequency is changed correctly under Win10 and Win11. Unfortunately, I don't know what could have changed since you installed 0.40.1, but so far it doesn't look like a QLog issue. Please, verify also your current Rig profile TX/RX offsets. Even though your rig is not connected, offsets can in some cases affect the frequency.

Can you send me from which version you upgraded?

DL2XT, 2024-12-19:
I upgraded from `v0.39.0` to `v0.40.1`. 

The active rig have no offset. I also tried it with a new one. No change... I. don´t know why this is Happening.

DL2XT, 2024-12-19:
it happens on both Versions. i downgraded and still the same behavior.
I think it is something on my windows Settings, but no freaking idea what it could be...

DL2XT, 2024-12-19:
I figured out if I enter `1.45` it will be converted to `145.000`. 
I also checked the registry entries, but Nothing found what could be related to this.

Any ideas?

aa5sh, 2024-12-19:
Are you on Windows 11 you can go to "Time & Language > Language & Region > Administrative Language Settings > Formats tab > Additional Settings > Numbers tab" now I'm in the US so not sure how different yours may supposed to be but check this out.  Here is what mine looks like:  
[image]

This is definitely a weird one.  There is an option at the bottom to reset to defaults.

DL2XT, 2024-12-19:
@aa5sh that fixed my issue! a huge thanks!!

aa5sh, 2024-12-19:
Glad you got it sorted.

DL2XT, 2024-12-20:
I tested some further and now it is working like a charm, like before! 
My decimal separator is now a comma (`,`), which is actually the correct one for me with an german layout and so on. 
However, I actually did not changed that :D

But a huge thanks for all participants! I really love `QLog`, it is a really great logging software!

closed 2024-12-20 by DL2XT

foldynl, 2024-12-20:
@aa5sh  Great insight