foldynl/QLog #559: Problems uploading to QRZ.com

Skittlebrau, 2024-12-30:
When I try to upload contacts to QRZ, only a few are uploaded and then a I receive a "connection closed" error.  I have reached out to QRZ support, but they think it might be an issue on my end, so I thought I'd check in here as well.  I have not had issues uploading things elsewhere and my internet is 2gbps and generally reliable elsewhere, so I would be surprised if it's a basic connectivity issue.  QRZ support mentioned something about an API rate limit, so perhaps one possibility is that QLog is exceeding that?

aa5sh, 2024-12-30:
How many contacts are you trying to upload and what do you mean by a few?  Just trying to get a clearer picture.  I know QLog does upload them one at a time so I'd be curious about their rate limit.  I've not experienced this error but I've never tried to upload more than 50 or so at a time at most.  If they have a limit be good to know what it is in case there needs to be something to send it bursts or add a delay ever so many you know.

aa5sh, 2024-12-30:
@foldynl In looking at https://www.qrz.com/docs/logbook/QRZLogbookAPI.html it talks about needing a user-agent header or it could be limited.  I may be missing it but don't see where one is being sent.  Should a line something like this be added to QRZ::actionInsert

`request.setRawHeader("User-Agent", "QLog/0.40 <OK1MLG>");`

Skittlebrau, 2024-12-31:
I have thousands to upload as I have just recently subscribed to QRZ and have not uploaded before.  No issues sending large numbers to LOTW.

foldynl, 2025-01-01:
@aa5sh  . Many thanks for PR. You're right. According to the API documentation, there should be a user-agent info. I would just change your PR to include an option for "widely distributed scripts".

aa5sh, 2025-01-01:
I guess the way I read it was for the “widely distributed scripts” was to use the application name and version.  But it is also had the callsign which is why I put yours but in looking again it may not be needed since it has the application name first.

Skittlebrau, 2025-01-14:
Whatever issue I am having this did not seem to help, as 0.41.0 is having the same behavior.  Is there any logging or something I can turn on?  I had things building on Windows at some point too, but I had to make a real mess of things locally to make it work (I am not at all familiar with Qt and it's build system).  I see there are new Windows build instructions to try as well.  But anyway, I might be able to experiment locally if I can get it building again.  I could also give the QRZ upload a shot under Linux if you think there's value in trying that, though it doesn't seem that likely to be platform-dependent?

aa5sh, 2025-01-14:
I can’t remember if we talked about this, but you could always export you log to ADI file and just manually upload it to QRZ then bulk edit that field in your log to a ‘Y’ for all QSOs.  Then it may be manageable going forward.

Skittlebrau, 2025-01-14:
I have so far intentionally avoided doing that so I could test any changes to see if they worked, but yeah that might work til the next contest dump comes in, at which point I'd probably have to go to adif again.

aa5sh, 2025-01-14:
Yeah that makes sense. Only thing I would be curious of still if it constantly stops after x records.  They don’t say anything about the need to send in smallish groups though.  I know on downloading from QRZ the API specs say to do it in small batches of 100-250.  But for uploading they are done 1 at a time.  But obviously something is stopping it whether it is on the QRZ side or something in your internet.  I wonder for instance if your firewall has some sort of IPS/IDS that is kicking in thinking it is some sort DDOS since the records are sent one at a time really fast.  Lots of variables.  Especially since you are not getting a good error back from QRZ.

foldynl, 2025-01-14:
0.41 contains more verbose QRZ upload procedure. So if you run QLog under windows then set variable QT_LOGGING_RULES before executing QLog with debug switch as below

```
set QT_LOGGING_RULES=qlog.core.qrz.*=true
qlog.exe -d
```

perform QRZ upload. After error, exit qlog and send us the debug log in 

`C:/Users/<USER>/AppData/Local/hamradio/QLog/qlog_debug_yyyyMMddhhmmss.log`

Skittlebrau, 2025-01-15:
Log does not seem to be very verbose, but here it is:
`App: 0.41.0
QT: 6.6.3
OS: Windows 11 Version 23H2 x86_64 (windows)
SSL: OpenSSL 3.3.2 3 Sep 2024

21:08:10.901 [INFO    ] [0x5298] [qlog.ui.qrzdialog.runtime]                        QRZ.com Upload Error:  "Connection closed" [auto __cdecl QRZDialog::upload::<lambda_3>::operator ()(const class QString &) const:..\QLog\ui\QrzDialog.cpp:141]
`

Skittlebrau, 2025-01-15:
Whoops, I was in powershell and did the set wrong.  

```
App: 0.41.0
QT: 6.6.3
OS: Windows 11 Version 23H2 x86_64 (windows)
SSL: OpenSSL 3.3.2 3 Sep 2024

21:13:24.581 [DEBUG   ] [0x62c8] [qlog.core.qrz.function.entered]                   *** [__cdecl QRZ::QRZ(class QObject *):..\QLog\core\QRZ.cpp:30]
21:13:24.895 [DEBUG   ] [0x62c8] [qlog.core.qrz.function.entered]                   *** [__cdecl QRZ::QRZ(class QObject *):..\QLog\core\QRZ.cpp:30]
21:13:24.930 [DEBUG   ] [0x62c8] [qlog.core.qrz.function.entered]                   *** [__cdecl QRZ::QRZ(class QObject *):..\QLog\core\QRZ.cpp:30]
21:13:31.473 [DEBUG   ] [0x62c8] [qlog.core.qrz.function.entered]                   *** [const class QString __cdecl QRZ::getLogbookAPIKey(void):..\QLog\core\QRZ.cpp:191]
21:13:33.884 [DEBUG   ] [0x62c8] [qlog.core.qrz.function.entered]                   *** [__cdecl QRZ::QRZ(class QObject *):..\QLog\core\QRZ.cpp:30]
21:13:33.884 [DEBUG   ] [0x62c8] [qlog.core.qrz.function.entered]                   *** [void __cdecl QRZ::uploadContacts(const class QList<class QSqlRecord> &):..\QLog\core\QRZ.cpp:153]
21:13:33.885 [DEBUG   ] [0x62c8] [qlog.core.qrz.function.entered]                   *** [void __cdecl QRZ::uploadContact(const class QSqlRecord &):..\QLog\core\QRZ.cpp:105]
21:13:33.885 [DEBUG   ] [0x62c8] [qlog.core.qrz.function.entered]                   *** [void __cdecl QRZ::actionInsert(class QByteArray &,const class QString &):..\QLog\core\QRZ.cpp:123]
21:13:33.885 [DEBUG   ] [0x62c8] [qlog.core.qrz.function.entered]                   *** [const class QString __cdecl QRZ::getLogbookAPIKey(void):..\QLog\core\QRZ.cpp:191]
21:13:33.887 [DEBUG   ] [0x62c8] [qlog.core.qrz.runtime]                            QUrl("https://logbook.qrz.com/api") [void __cdecl QRZ::actionInsert(class QByteArray &,const class QString &):..\QLog\core\QRZ.cpp:140]
21:13:34.067 [DEBUG   ] [0x62c8] [qlog.core.qrz.function.entered]                   *** [void __cdecl QRZ::processReply(class QNetworkReply *):..\QLog\core\QRZ.cpp:290]
21:13:34.067 [DEBUG   ] [0x62c8] [qlog.core.qrz.runtime]                            Received Message Type:  "actionsInsert" [void __cdecl QRZ::processReply(class QNetworkReply *):..\QLog\core\QRZ.cpp:318]
21:13:34.067 [DEBUG   ] [0x62c8] [qlog.core.qrz.runtime]                            "COUNT=1&RESULT=OK&LOGID=1213679663" [void __cdecl QRZ::processReply(class QNetworkReply *):..\QLog\core\QRZ.cpp:511]
21:13:34.067 [DEBUG   ] [0x62c8] [qlog.core.qrz.function.entered]                   *** [class QMap<class QString,class QString> __cdecl QRZ::parseActionResponse(const class QString &) const:..\QLog\core\QRZ.cpp:548]
21:13:34.067 [DEBUG   ] [0x62c8] [qlog.core.qrz.function.parameters]                "COUNT=1&RESULT=OK&LOGID=1213679663" [class QMap<class QString,class QString> __cdecl QRZ::parseActionResponse(const class QString &) const:..\QLog\core\QRZ.cpp:550]
21:13:34.067 [DEBUG   ] [0x62c8] [qlog.core.qrz.runtime]                            Confirmed Upload for QSO Id  770 [void __cdecl QRZ::processReply(class QNetworkReply *):..\QLog\core\QRZ.cpp:519]
21:13:34.069 [DEBUG   ] [0x62c8] [qlog.core.qrz.function.entered]                   *** [void __cdecl QRZ::uploadContact(const class QSqlRecord &):..\QLog\core\QRZ.cpp:105]
21:13:34.069 [DEBUG   ] [0x62c8] [qlog.core.qrz.function.entered]                   *** [void __cdecl QRZ::actionInsert(class QByteArray &,const class QString &):..\QLog\core\QRZ.cpp:123]
21:13:34.069 [DEBUG   ] [0x62c8] [qlog.core.qrz.function.entered]                   *** [const class QString __cdecl QRZ::getLogbookAPIKey(void):..\QLog\core\QRZ.cpp:191]
21:13:34.070 [DEBUG   ] [0x62c8] [qlog.core.qrz.runtime]                            QUrl("https://logbook.qrz.com/api") [void __cdecl QRZ::actionInsert(class QByteArray &,const class QString &):..\QLog\core\QRZ.cpp:140]
21:13:34.232 [DEBUG   ] [0x62c8] [qlog.core.qrz.function.entered]                   *** [void __cdecl QRZ::processReply(class QNetworkReply *):..\QLog\core\QRZ.cpp:290]
21:13:34.232 [DEBUG   ] [0x62c8] [qlog.core.qrz.runtime]                            Received Message Type:  "actionsInsert" [void __cdecl QRZ::processReply(class QNetworkReply *):..\QLog\core\QRZ.cpp:318]
21:13:34.232 [DEBUG   ] [0x62c8] [qlog.core.qrz.runtime]                            "LOGID=1213679664&RESULT=OK&COUNT=1" [void __cdecl QRZ::processReply(class QNetworkReply *):..\QLog\core\QRZ.cpp:511]
21:13:34.232 [DEBUG   ] [0x62c8] [qlog.core.qrz.function.entered]                   *** [class QMap<class QString,class QString> __cdecl QRZ::parseActionResponse(const class QString &) const:..\QLog\core\QRZ.cpp:548]
21:13:34.232 [DEBUG   ] [0x62c8] [qlog.core.qrz.function.parameters]                "LOGID=1213679664&RESULT=OK&COUNT=1" [class QMap<class QString,class QString> __cdecl QRZ::parseActionResponse(const class QString &) const:..\QLog\core\QRZ.cpp:550]
21:13:34.232 [DEBUG   ] [0x62c8] [qlog.core.qrz.runtime]                            Confirmed Upload for QSO Id  771 [void __cdecl QRZ::processReply(class QNetworkReply *):..\QLog\core\QRZ.cpp:519]
21:13:34.233 [DEBUG   ] [0x62c8] [qlog.core.qrz.function.entered]                   *** [void __cdecl QRZ::uploadContact(const class QSqlRecord &):..\QLog\core\QRZ.cpp:105]
21:13:34.234 [DEBUG   ] [0x62c8] [qlog.core.qrz.function.entered]                   *** [void __cdecl QRZ::actionInsert(class QByteArray &,const class QString &):..\QLog\core\QRZ.cpp:123]
[... cut ...]
21:13:53.655 [DEBUG   ] [0x62c8] [qlog.core.qrz.runtime]                            QRZ.com error "Connection closed" [void __cdecl QRZ::processReply(class QNetworkReply *):..\QLog\core\QRZ.cpp:302]
21:13:53.655 [DEBUG   ] [0x62c8] [qlog.core.qrz.runtime]                            HTTP Status Code 0 [void __cdecl QRZ::processReply(class QNetworkReply *):..\QLog\core\QRZ.cpp:303]
21:13:53.659 [INFO    ] [0x62c8] [qlog.ui.qrzdialog.runtime]                        QRZ.com Upload Error:  "Connection closed" [auto __cdecl QRZDialog::upload::<lambda_3>::operator ()(const class QString &) const:..\QLog\ui\QrzDialog.cpp:141]
```

foldynl, 2025-01-15:
Unfortunately, I don't know how to help you. According to the logs, it appears that QRZ.com terminated the connection at the network level. Why this happened, I cannot say.

From the output, it's clear that you sent three messages, and the issue occurred with the fourth. I tried sending ten messages myself and successfully sent two batches with 10 QSOs without any problems. It seems to be some sort of network error, but this would need to be debugged at the TCP level directly on your computer.

Skittlebrau, 2025-01-15:
I took a look at Wireshark and there does seem to be weird TCP retries that occur on the problem API call.  I tried a ton of stuff locally including using wifi instead of the usual wired connection, ISP-provided router instead of my usual Opnsense router, various router settings -- none of which had any effect.  I tried various ways of detecting packet loss problems, but everything came out good with 0% packet loss.  I pored over the Wireshark logs a bit more.  It's not my area of expertise by any means, but I did notice something a bit strange (consistent over a bunch of tests), which is that the TLS "Client Hello" from QLog to QRZ is showing as "TLSv1.3" for every successful API call and "TLSv1.2" for the failed ones.  The Client Hello packet does still advertise both 1.2 and 1.3 in the supported_versions, however.  This might be a total red herring, but posting it just in case it suggests any clues to anyone else (like some openssl + win11 thing?)

Skittlebrau, 2025-01-15:
Under linux on same computer it acts similarly, except that it just "hangs" instead of getting the connection closed error.

`
14:44:52.145 [DEBUG   ] [0x70a192b6fa40] [qlog.core.qrz.function.entered]                   *** [static const QString QRZ::getLogbookAPIKey():core/QRZ.cpp:191]
14:44:54.735 [DEBUG   ] [0x70a192b6fa40] [qlog.core.qrz.function.entered]                   *** [QRZ::QRZ(QObject*):core/QRZ.cpp:30]
14:44:54.735 [DEBUG   ] [0x70a192b6fa40] [qlog.core.qrz.function.entered]                   *** [void QRZ::uploadContacts(const QList<QSqlRecord>&):core/QRZ.cpp:153]
14:44:54.735 [DEBUG   ] [0x70a192b6fa40] [qlog.core.qrz.function.entered]                   *** [void QRZ::uploadContact(const QSqlRecord&):core/QRZ.cpp:105]
14:44:54.736 [DEBUG   ] [0x70a192b6fa40] [qlog.core.qrz.function.entered]                   *** [void QRZ::actionInsert(QByteArray&, const QString&):core/QRZ.cpp:123]
14:44:54.736 [DEBUG   ] [0x70a192b6fa40] [qlog.core.qrz.function.entered]                   *** [static const QString QRZ::getLogbookAPIKey():core/QRZ.cpp:191]
14:44:54.738 [DEBUG   ] [0x70a192b6fa40] [qlog.core.qrz.runtime]                            QUrl("https://logbook.qrz.com/api") [void QRZ::actionInsert(QByteArray&, const QString&):core/QRZ.cpp:140]
14:44:54.991 [DEBUG   ] [0x70a192b6fa40] [qlog.core.qrz.function.entered]                   *** [void QRZ::processReply(QNetworkReply*):core/QRZ.cpp:290]
14:44:54.991 [DEBUG   ] [0x70a192b6fa40] [qlog.core.qrz.runtime]                            Received Message Type:  "actionsInsert" [void QRZ::processReply(QNetworkReply*):core/QRZ.cpp:318]
14:44:54.991 [DEBUG   ] [0x70a192b6fa40] [qlog.core.qrz.runtime]                            "LOGID=1214088767&RESULT=OK&COUNT=1" [void QRZ::processReply(QNetworkReply*):core/QRZ.cpp:511]
14:44:54.991 [DEBUG   ] [0x70a192b6fa40] [qlog.core.qrz.function.entered]                   *** [QMap<QString, QString> QRZ::parseActionResponse(const QString&) const:core/QRZ.cpp:548]
14:44:54.991 [DEBUG   ] [0x70a192b6fa40] [qlog.core.qrz.function.parameters]                "LOGID=1214088767&RESULT=OK&COUNT=1" [QMap<QString, QString> QRZ::parseActionResponse(const QString&) const:core/QRZ.cpp:550]
14:44:54.991 [DEBUG   ] [0x70a192b6fa40] [qlog.core.qrz.runtime]                            Confirmed Upload for QSO Id  1151 [void QRZ::processReply(QNetworkReply*):core/QRZ.cpp:519]
14:44:54.997 [DEBUG   ] [0x70a192b6fa40] [qlog.core.qrz.function.entered]                   *** [void QRZ::uploadContact(const QSqlRecord&):core/QRZ.cpp:105]
14:44:54.998 [DEBUG   ] [0x70a192b6fa40] [qlog.core.qrz.function.entered]                   *** [void QRZ::actionInsert(QByteArray&, const QString&):core/QRZ.cpp:123]
14:44:54.998 [DEBUG   ] [0x70a192b6fa40] [qlog.core.qrz.function.entered]                   *** [static const QString QRZ::getLogbookAPIKey():core/QRZ.cpp:191]
14:44:54.999 [DEBUG   ] [0x70a192b6fa40] [qlog.core.qrz.runtime]                            QUrl("https://logbook.qrz.com/api") [void QRZ::actionInsert(QByteArray&, const QString&):core/QRZ.cpp:140]
14:44:55.222 [DEBUG   ] [0x70a192b6fa40] [qlog.core.qrz.function.entered]                   *** [void QRZ::processReply(QNetworkReply*):core/QRZ.cpp:290]
14:44:55.222 [DEBUG   ] [0x70a192b6fa40] [qlog.core.qrz.runtime]                            Received Message Type:  "actionsInsert" [void QRZ::processReply(QNetworkReply*):core/QRZ.cpp:318]
14:44:55.222 [DEBUG   ] [0x70a192b6fa40] [qlog.core.qrz.runtime]                            "RESULT=OK&LOGID=1214088768&COUNT=1" [void QRZ::processReply(QNetworkReply*):core/QRZ.cpp:511]
14:44:55.222 [DEBUG   ] [0x70a192b6fa40] [qlog.core.qrz.function.entered]                   *** [QMap<QString, QString> QRZ::parseActionResponse(const QString&) const:core/QRZ.cpp:548]
14:44:55.222 [DEBUG   ] [0x70a192b6fa40] [qlog.core.qrz.function.parameters]                "RESULT=OK&LOGID=1214088768&COUNT=1" [QMap<QString, QString> QRZ::parseActionResponse(const QString&) const:core/QRZ.cpp:550]
14:44:55.222 [DEBUG   ] [0x70a192b6fa40] [qlog.core.qrz.runtime]                            Confirmed Upload for QSO Id  1152 [void QRZ::processReply(QNetworkReply*):core/QRZ.cpp:519]
14:44:55.225 [DEBUG   ] [0x70a192b6fa40] [qlog.core.qrz.function.entered]                   *** [void QRZ::uploadContact(const QSqlRecord&):core/QRZ.cpp:105]
14:44:55.226 [DEBUG   ] [0x70a192b6fa40] [qlog.core.qrz.function.entered]                   *** [void QRZ::actionInsert(QByteArray&, const QString&):core/QRZ.cpp:123]
14:44:55.226 [DEBUG   ] [0x70a192b6fa40] [qlog.core.qrz.function.entered]                   *** [static const QString QRZ::getLogbookAPIKey():core/QRZ.cpp:191]
14:44:55.228 [DEBUG   ] [0x70a192b6fa40] [qlog.core.qrz.runtime]                            QUrl("https://logbook.qrz.com/api") [void QRZ::actionInsert(QByteArray&, const QString&):core/QRZ.cpp:140]
14:44:55.433 [DEBUG   ] [0x70a192b6fa40] [qlog.core.qrz.function.entered]                   *** [void QRZ::processReply(QNetworkReply*):core/QRZ.cpp:290]
14:44:55.433 [DEBUG   ] [0x70a192b6fa40] [qlog.core.qrz.runtime]                            Received Message Type:  "actionsInsert" [void QRZ::processReply(QNetworkReply*):core/QRZ.cpp:318]
14:44:55.433 [DEBUG   ] [0x70a192b6fa40] [qlog.core.qrz.runtime]                            "RESULT=OK&LOGID=1214088770&COUNT=1" [void QRZ::processReply(QNetworkReply*):core/QRZ.cpp:511]
[... cut ...]
14:47:47.109 [DEBUG   ] [0x79228e553a40] [qlog.core.qrz.function.entered]                   *** [void QRZ::actionInsert(QByteArray&, const QString&):core/QRZ.cpp:123]
14:47:47.109 [DEBUG   ] [0x79228e553a40] [qlog.core.qrz.function.entered]                   *** [static const QString QRZ::getLogbookAPIKey():core/QRZ.cpp:191]
14:47:47.111 [DEBUG   ] [0x79228e553a40] [qlog.core.qrz.runtime]                            QUrl("https://logbook.qrz.com/api") [void QRZ::actionInsert(QByteArray&, const QString&):core/QRZ.cpp:140]
`

foldynl, 2025-01-16:
>>> which is that the TLS "Client Hello" from QLog to QRZ is showing as "TLSv1.3" for every successful API call and "TLSv1.2" for the failed ones.

And IP addresses are the same for both these TLS1.3 and TLS1.2 requests ? I don't know. Just brainstorming

>> Under linux on the same computer it acts similarly, except that it just "hangs" 

Based on the debug log, it is very strange because it is an asynchronous function. There is no reason to freeze here.
What installation package are you using under Linux DEB/RPM/Flatpak/own compilation?

Skittlebrau, 2025-01-16:
> > > > which is that the TLS "Client Hello" from QLog to QRZ is showing as "TLSv1.3" for every successful API call and "TLSv1.2" for the failed ones.
> 
> And IP addresses are the same for both these TLS1.3 and TLS1.2 requests ? I don't know. Just brainstorming
> 
Yeah IP addresses were the same, but of course it might be different servers behind a load balancer or something.

> > > Under linux on the same computer it acts similarly, except that it just "hangs"
> 
> Based on the debug log, it is very strange because it is an asynchronous function. There is no reason to freeze here. What installation package are you using under Linux DEB/RPM/Flatpak/own compilation?

Just to re-clarify, it did not literally hang, but it just stopped sending more QSOs, but didn't fail with an error either.  It could be stopped with the cancel button.  I am using the .deb (I am pretty sure) on Kubuntu.

I can dig into this more on my own if you are tired of helping me speculate on what be going on.  I have a ticket in with QRZ too, though they are being slow at responding.

aa5sh, 2025-01-16:
In looking in the QT documents it looks like the request could be forced to use TLS 1.3 by doing something like this:
￼

But I have no idea if that would cause other issues.  Just weird that it would be changing for different requests.

aa5sh, 2025-01-16:
[image]

Not sure why it looks like it didn't load the image from email.

foldynl, 2025-01-17:
I don't think this is the solution. I will be interesting to test it, but the TLS version can be enforced by the server, not the client. Qt manages the connection pool itself. It has quite a robust logic and automatically parallelizes queries. If one connection switched to TLS 1.2, it means the server requested it. And that’s exactly what I was asking—whether this connection came from a different IP address or the same one as the TLS 1.3 connection. That was the point of my question.

Skittlebrau, 2025-01-28:
FWIW, QRZ said they looked at my logs and that I am being rate-limited by their side.  I tried to ask what the rate limit is (since best I can tell that's not documented), but they have not replied with anything yet.  I went to check to see if I just missed an email but they had closed my ticket without response.  I tried reopening it, so hopefully we will hear something.

Skittlebrau, 2025-01-30:
Well now they think I'm not hitting the rate limit, which is not really documented in the first place, but is heuristic.  Mentioned 60 calls/minute but also said they had no issues uploading at 200 calls/min from QLog in their own tests.  So perhaps it is some strange thing between me and qrz.com after all, despite other sites seemingly working fine.

aa5sh, 2025-01-30:
Would you have the option to try from a hotspot or a different internet
connection?  Doesn’t really make sense as to what is happening.