foldynl/QLog #563: Two QSOs confirmed in LoTW with close timing cannot be matched.

xiaoxi654, 2025-01-03:
I have two QSOs with a time difference of about ten minutes. After uploading them to LoTW, QLog indicates a mismatch when retrieving the QSL information from LoTW.
[image]
[image]
[image]

aa5sh, 2025-01-03:
Yeah the way the confirmation logic works today is it checks for matches within a 30 minute (I think) window.  It will exception out the QSL if it matches more than one log entry, since it may be ambiguous.  So typically when I have those I just manually set the LoTW Recieved flag to a Yes.

foldynl, 2025-01-04:
it's necessary to look at this problem from another perspective. Yes, your issue lies in the fact that QLog "did not recognize" QSOs made within a 10-minute timeframe. However, there is another scenario: operators often do not have the "same time". That’s why services like eQSL, LoTW, and others tolerate a time difference (-+30minute). 

QLog also has this implemented. This means that if a QSO with the same callsign, band, and mode falls within this tolerance window, it is treated as a single QSO, as in your case. In such situations, I recommend following AA5SH advice – perform a manual update of the record.

what could be worth modifying is to modify the matching rule from callsign, band, mode to callsign, band, mode, sat_name. I think that is an acceptable solution to this problem,

xiaoxi654, 2025-01-05:
> it's necessary to look at this problem from another perspective. Yes, your issue lies in the fact that QLog "did not recognize" QSOs made within a 10-minute timeframe. However, there is another [...]

Ah, that’s reasonable. Adding the satellite name to the matching rules should solve this problem. Thanks!

commit 9ff5fdba (2025-01-04): Fixed #563 - Extended QSL/Import Dupe matching rule from Callsign, Band, Mode, Time to Callsign, Band, Mode, Time, Sat Name

This rule more closely reflects the LoTW matching rules.
Since it is possible to make a QSO on a Sat 'A' and within a few minutes on Sat 'B' with the same Callsign
closed 2025-01-11 by foldynl