foldynl/QLog #565: Enhancement: Update QTH and Grid when updating POTA

vazquezrodjoel, 2025-01-04:
To Update QTH and Grid when updating POTA field on an existing/saved contact.

If I edit and finished/saved contact with the POTA number, I have to manually update the QTH and Grid.

This is not a real contact that need update, but just a sample:
[image]

foldynl, 2025-01-06:
Originally, this form was designed to give the operator freedom, which is why there are no automatic suggestions. However, to validate data consistency, there are validations in place that indicate what Qlog considers to be the correct value, leaving it up to the operator's discretion to decide which value to use. 

I believe this approach should be preserved, so I would keep this form without automatic suggestions.

vazquezrodjoel, 2025-01-06:
I see your point of view, I'm just thinking of those times we would like to update all at once based on the missed POTA from like FT8 contacts (not sure if IOTA works like POTA in that sense, but maybe could be the same.  So maybe would consider adding a "click-to-update" option on those messages/warning?  
Like "I notice that the update does not correlate to the saved information, would you like to expand and auto-update the QTH and Grid to the matching value?   Yes/No"...