foldynl/QLog #569: Edit QSO complains about ITU zone when it shouldn't

quantalume, 2025-01-10:
I have two locations that I operate from in Texas and New Mexico. If I edit a QSO, it will complain about the ITU zone for my station not being correct. How does it make this determination? The gridsquare entry is correct.
[image]

foldynl, 2025-01-10:
ITUz is derived from the callsign and the DXCC Entity database (CTY-file). Unfortunately, for large countries like the USA, this method is not accurate. The yellow warning is more about to display an alert the operator to check if it is correct.

quantalume, 2025-01-11:
Thank you for responding quickly. I looked through the code and found where you are getting this information from https://www.country-files.com/bigcty/cty.csv. 

It is not a big deal to me. An easy solution is to ignore the "error" if the fields (including gridsquare) are in agreement with the saved station profile. Perhaps only check them when the station profile is added or modified?

Thanks for all your work on this code.

foldynl, 2025-01-11:
if ITU/CQ should  be checked based on the provided information, the task would be entirely different. Here’s why:

- You would need to precisely define the boundaries of ITU/CQ zones. This is not as straightforward as it may seem at first glance. Once that's established, the zones can be determined based on the grid.
- If grid is unknown but only the QTH (city) is known, it would require updating the location of all cities worldwide. The world isn’t limited to US territories ;-)

But I’m open to any pull request that aims to improve this.

aa5sh, 2025-01-21:
Wonder if it would make sense to maybe reword the message and that may limit any confusion.  Possibly like 'Yellow marked may have incorrect data please confirm' or something along those lines.