foldynl/QLog #570: Got warning while compiling Release 0.41 (missing flags.qrc)

vazquezrodjoel, 2025-01-11:
commands  executed:
`git clone https://github.com/foldynl/QLog.git .`
`PREFIX=/opt/qlog qmake QLog.pro`
Got:
`WARNING: Failure to find: res/flags/flags.qrc
/usr/lib/qt5/bin/rcc: File does not exist 'res/flags/flags.qrc'`

When I looked at res/flags/ folder, it was empty.
Source Code zip folder also seems empty.

I downloaded them manually, repeated the commands, and it went fine.

[image]

is that normal/expected behavior??

aa5sh, 2025-01-11:
I've seen this before.  I've just went to github and downloaded the missing
files and saved them. Not sure why they don't get included in the source
package.

foldynl, 2025-01-11:
I was mentioned many times. Please follow instructions in [README](https://github.com/foldynl/qlog?tab=readme-ov-file#linux-1). Git repo contains a flag module. Therefore it is needed to download it:

git clone --recurse-submodules https://github.com/foldynl/QLog.git

aa5sh, 2025-01-11:
I have somehow missed this.  At this point I use the github desktop app (I
know I need to learn command line - my son tells me all the time), so I
need to figure out how to do the recourse-submodules.

Thanks for this.

Michael

closed 2025-04-05 by foldynl