foldynl/QLog #571: complile issue on Debian 12

Murzim2, 2025-01-11:
Problem compiling latest release 11/01/2024 for Debian (QT6)
Stopped with the following error.
 rig/drivers/HamlibRigDrv.cpp:57:5: error: no declaration matches ‘int HamlibRigDrv::addRig(rig_caps*, void*)’
   57 | int HamlibRigDrv::addRig(rig_caps *caps, void* data)
      |     ^~~~~~~~~~~~
In file included from rig/drivers/HamlibRigDrv.cpp:4:
rig/drivers/HamlibRigDrv.h:49:16: note: candidate is: ‘static int HamlibRigDrv::addRig(const rig_caps*, void*)’
   49 |     static int addRig(const rig_caps *caps, void* data);
      |                ^~~~~~
rig/drivers/HamlibRigDrv.h:11:7: note: ‘class HamlibRigDrv’ defined here
   11 | class HamlibRigDrv : public GenericRigDrv
      |       ^~~~~~~~~~~~
make: *** [Makefile:4536: HamlibRigDrv.o] Error 1

vazquezrodjoel, 2025-01-11:
I got a similar error, I started from scratch (like redownload everything).  then it worked.

foldynl, 2025-01-11:
Ah...OK...please, do not use master branch of Hamlib repo but Hamlib-4.6 branch. from the perspective of an operator you won't know the difference. But there is an API difference. I handled it incorrectly in QLog.

foldynl, 2025-01-11:
or you can apply this patch
```
diff --git a/rig/drivers/HamlibRigDrv.cpp b/rig/drivers/HamlibRigDrv.cpp
index 95bc031..7b4a01e 100644
--- a/rig/drivers/HamlibRigDrv.cpp
+++ b/rig/drivers/HamlibRigDrv.cpp
@@ -53,7 +53,7 @@ QList<QPair<QString, QString> > HamlibRigDrv::getPTTTypeList()
     return ret;
 }
 
-#if ( HAMLIBVERSION_MAJOR >= 4 && HAMLIBVERSION_MINOR >= 6 )
+#if ( HAMLIBVERSION_MAJOR == 4 && HAMLIBVERSION_MINOR == 6 && HAMLIBVERSION_PATCH == 0  )
 int HamlibRigDrv::addRig(rig_caps *caps, void* data)
 #else
 int HamlibRigDrv::addRig(const rig_caps *caps, void* data)
```

and use qmake parameter HAMLIBVERSION_PATCH = 1

commit 97f8a376 (2025-01-21): Fixed #571 - Hamlib API changes between 4.6.0 and 4.6.1
commit ee39df0a (2025-01-21): Fixed #571 - HamlibDrv: Used newer API to get information about Rig