foldynl/QLog #577: The QSO ID count will only increase, not decrease

BI6NSL, 2025-01-20:
I deleted some QSO logs for some reason, when I corrected the qso data and imported it back again (Qlog), I found that the QSO ID was cumulative, and did not subtract the part of the QSO ID that I deleted.For example, 1+1=2,2+1=3,3-1+1=5.What am I supposed to do?

[image]

[image]

[image]

aa5sh, 2025-01-20:
The qso_id column is an incrementing unique key entry for the contacts table in the database.  It just goes up by one any time a new log entry is added.  So if you delete an entry it will just be a gap so to speak if you sort them by that number.  It will not reuse them.  Not sure if I explained that well.

Michael, AA5SH

foldynl, 2025-01-21:
Michael, you are right. QSO ID is not a counter but an unique number. This number is used by QLog for faster manipulations. It should be "hidden" for the average operator.

BI6NSL, 2025-01-21:
I think I get it. Thank you.

closed 2025-01-21 by foldynl