foldynl/QLog #578: Request to add [Select All QSO] option

BI6NSL, 2025-01-20:
Request to add [Select All QSO] option

Request to add the [Select All QSO] option to work with the user filter, or use it alone, each batch operation of some qso requires a long copy with the mouse, it would be nice to have a [Select all], most of the time you need to export some qso for specific needs.
It takes a long scroll to select them all

My English is not very good. Please forgive me for using Youdao Translation to communicate with you. I hope the translation software can fully understand what I want to express

[image]

BI6NSL, 2025-01-20:
Thank you. I hope you can think about it. Please!

aa5sh, 2025-01-20:
This was there but was removed I think in v.0.40.0.  You can create a user filter then export all entries that match the user filter.  That is what I typically do.

Michael

foldynl, 2025-01-21:
I am aware of this 'imperfection.' Hopefully a way can be found to implement something similar while maintaining processing speed. It is important to know that this is not just about 'Select All' but rather about what the operator intends to do with the selection. There are operators here who have 100k or more QSOs.

BI6NSL, 2025-01-21:
Create a user filter, and then export all entries that match the user filter.
This export method exports the matching data of the user filter instead of ALL data. Is my understanding correct?

BI6NSL, 2025-01-21:
After each export, I have to use ADIF Master import and export again for internal layout problems before they become the format shown in the following figure

[image]

aa5sh, 2025-01-21:
Yeah I have a few exports for POTA or just FT4/FT8 entries to update my WSJT-X adi file.  

[image]

Why do you reformat so it is on same line?  It really shouldn't matter, just cosmetic, unless other applications don't process the file properly.  Just curious.

Michael, AA5SH

BI6NSL, 2025-01-21:
Thanks Michael AA5SH, I don't think I've studied Qlog enough