foldynl/QLog #581: Request to add DX Labs Commander Rig Support

krossphire, 2025-01-25:
I've had my rig settings for my FTDX-5000 working well in QLog, but lately, it seems like nothing wants to work. A rig connection that has always been really solid for me in the DL Labs suite Commander. It would be cool to see Commander offered as a Rig connection setting option such as what WSJT-X offer. Thanks and 73 de K5PO

krossphire, 2025-01-26:
I'm finding that for my FTDX-5000, WSJT-X with Hamlib "4.5.5 Apr 04" will keep a constant connection while Qlog with Hamlib "20241118.11" won't seem the connect with the same settings.

foldynl, 2025-01-26:
just for reference, Commander message specification. http://www.dxlabsuite.com/commander/Commander%20TCPIP%20Messages.pdf

krossphire, 2025-01-26:
Also finding that Ham Radio Deluxe rig connection is working consistently.

aa5sh, 2025-01-26:
Have you tried with omnirig?

Michael, AA5SH

krossphire, 2025-01-26:
> Have you tried with omnirig?
> 
> Michael, AA5SH

I can connect with OmniRig V1 and QLog will communicate with it, however, if I open a second application on the same VSPE Virtual Splitter, the frequency being captured by QLog suddenly becomes inaccurate. 

Example, I'm on 28.444. Once a second application using the same virtual splitter as OmniRig is opened, the frequency in QLOg reflects 11.92844 MHz (but I'm still on 28.444)

foldynl, 2025-01-26:
It's strange that many users of FTDX have issues with all sorts of things in QLog. Maybe it would be good if at least one operator dive into analyzing the problem and sends a pull request to fix it.

aa5sh, 2025-01-31:
I am using a FTDX-101D that I recently switched to from a Flex 6400M.  I initially connected straight to QLog or WSJT-X and had no issues.  Now I launch a rigctld instance using 'rigctld -m 1040 -r /dev/cu.SLAB_USBtoUART' on my Mac.  Only issue I have is the need to adjust the port at times if I disconnect the USB cable for whatever reason.  I then will use rigctl.net rig in QLog and WSJTX.  The only thing I notice - and I think it is a hamlib issue - is if on my rig I select the sub vfo as the tuning knob control QLog and WSJT starts to display the sub vfo frequency instead of Main.  But that is really minor in my opinion.

Have you tried to have the rig talk directly to QLog not going through any VSPEs or anything?

aa5sh, 2025-01-31:
> I'm finding that for my FTDX-5000, WSJT-X with Hamlib "4.5.5 Apr 04" will keep a constant connection while Qlog with Hamlib "20241118.11" won't seem to connect with the same settings.

Can you send a screenshot of your WSJTX settings and your QLog settings so I can see what is working in WSJTX?

krossphire, 2025-02-07:
Apologies. I've got the FTDX-5000 off the desk and boxed up now. Running a Flex 6400 now.

aa5sh, 2025-02-28:
Sorry I missed the last question.  I use my Flex 6400 with QLog over TCP.  These are the settings I use.  HamLib recently added direct support where you connect to the rig instead of the client but when I tried that it was finicky with QLog or WSJT.  But I haven't tried it in a while.

[image]

But with these settings I never have a problem.

aa5sh, 2025-03-15:
Glad you got it working well.